plerup/makeEspArduino #2: `TOOLS_BIN` not correct when using ESP8266 core installed by Arduino Boards Manager

Bitbored, 2016-03-03:
The Arduino Boards Manager installs tools under `{TOOL_PATH}/{version}/{TOOL}`, e.g `esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2` however the ESP8266 [Arduino core get.py script](https://github.com/esp8266/Arduino/blob/2.1.0/tools/get.py) installs them directly under `{TOOL_PATH}/{TOOL}`, e.g `esp8266/tools/xtensa-lx106-elf`.

This means the makefile is not able to find the required tools for compiling/uploading the sketch when `ESP_ROOT` is set to the path where the Arduino Boards Manager installs the ESP8266 package.

This is mostly a problem with [ESP8266 core for Arduino](https://github.com/esp8266/Arduino) script not installing the tools in the correct location, however IMO the way the Arduino Boards Manager installs tools should be considered the correct way, and should be supported by this makefile.

plerup, 2016-03-06:
Well, the main intent for this makefile is to use the ESP8266-Arduino framework from a git in order to get a good stringent way of determine the version used in the build.

Bitbored, 2016-03-07:
Yeah, I get that. I mainly want to make clear that this makefile can not be used when the ESP8266 core, and thus the required tools, have been installed by the Arduino Boards manager.

It would be interesting to have this makefile working regardless of how ESP8266-Arduino has been installed, but I think the proper way to do this is to have the get.py script install the tools in the canonical locations. Until then, this issue is should inform people they have to use ESP8266-Arduino installed using git.

plerup, 2016-08-17:
Fixed in the latest commit

closed 2016-08-27 by plerup