plerup/makeEspArduino #7: compile of Blynk example fail

gpancot, 2016-05-06:
compile of ESP8266_Standalone sample from Blynk lib fail:

zero@fight ~/src/arduino/esp8266/ESP8266_Standalone $ make -f makeEspArduino.mk
main.cpp
In file included from /home/zero/src/arduino/libraries/Blynk/linux/main.cpp:17:0:
/home/zero/src/arduino/libraries/Blynk/linux/BlynkSocket.h:13:24: fatal error: sys/socket.h: No such file or directory
 #include \<sys/socket.h\>

gpancot, 2016-05-06:
solved, for now, but I do not like:

+
+USER_INCLUDE ?= $(HOME)/src/arduino/libraries/Blynk

+LIBS ?= \
+               $(ESP_LIBS)/ESP8266WiFi \
+               $(HOME)/src/arduino/libraries/Blynk/Blynk \
+               $(HOME)/src/arduino/libraries/Blynk/utility \
+

-INCLUDE_DIRS += $(SDK_ROOT)/include $(SDK_ROOT)/lwip/include $(CORE_DIR) $(ESP_ROOT)/variants/generic $(OBJ_DIR)
+INCLUDE_DIRS += $(SDK_ROOT)/include $(SDK_ROOT)/lwip/include $(CORE_DIR) $(ESP_ROOT)/variants/generic $(OBJ_DIR) $(USER_INCLUDE)

plerup, 2016-06-12:
Well the intent for cases like this was just to add an initial:

INCLUDE_DIRS = $(HOME)/src/arduino/libraries/Blynk

That's the reason for the +=

gpancot, 2016-06-23:
simply it works! ;-)
thanks.

closed 2016-07-25 by plerup