plerup/makeEspArduino #13: collect2: error: ld returned 1 exit status

diazleonardo, 2016-07-15:
I am trying to compile the simplest ino file 
```c    
#include <Arduino.h>
void setup() {
    pinMode(D0, OUTPUT);
}
void loop() {
}
```

but fails with the error shown in the subject and,

    /tmp/test/obj/core.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':
    /home/leonardo/esp8266/cores/esp8266/core_esp8266_main.cpp:110: undefined reference to `printf'

I tried adding `-ld` as mentioned in other cross-compile forums, but it doesn't help

can anybody help? Ideas?

plerup, 2016-07-25:
Please describe your setup more in detail

igrr, 2016-07-28:
I have it failing as well (although not exactly with the same message) with the latest git version of the ESP8266 core.
Some time ago we have added a few new flags, which are not present in makefile.

Here's the list of changes to platform.txt:
https://github.com/esp8266/Arduino/commits/292912402891da1305c629b698cc106fe9f8e5d7/platform.txt

igrr, 2016-07-28:
Here's the summary of changes required. This probably needs to be wrapped with some grep magic similar to what was done for libstdc++, to be backwards compatible.

```patch
From aeff3bbec5e8bbeb811ca22aaed5ec060ce3cb06 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Thu, 28 Jul 2016 20:10:14 +0800
Subject: [PATCH] Update cflags and ldflags for 2.3.0+

---
 makeEspArduino.mk | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)

diff --git a/makeEspArduino.mk b/makeEspArduino.mk
index cb48a75..aa9a03b 100644
--- a/makeEspArduino.mk
+++ b/makeEspArduino.mk
@@ -86,14 +86,14 @@ ESP_TOOL = $(TOOLS_ROOT)/esptool/esptool
 OTA_TOOL = $(TOOLS_ROOT)/espota.py
 HTTP_TOOL = curl
 
-INCLUDE_DIRS += $(SDK_ROOT)/include $(SDK_ROOT)/lwip/include $(CORE_DIR) $(ESP_ROOT)/variants/generic $(OBJ_DIR)
+INCLUDE_DIRS += $(SDK_ROOT)/include $(SDK_ROOT)/lwip/include $(SDK_ROOT)/libc/xtensa-lx106-elf/include $(CORE_DIR) $(ESP_ROOT)/variants/generic $(OBJ_DIR)
 C_DEFINES = -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266
 C_INCLUDES = $(foreach dir,$(INCLUDE_DIRS) $(USER_DIRS),-I$(dir))
 C_FLAGS ?= -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections
 CPP_FLAGS ?= -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections
 S_FLAGS ?= -c -g -x assembler-with-cpp -MMD
-LD_FLAGS ?= -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -L$(SDK_ROOT)/lib -L$(SDK_ROOT)/ld -T$(FLASH_LAYOUT) -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy
-LD_STD_LIBS ?= -lm -lgcc -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto -laxtls
+LD_FLAGS ?= -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static -L$(SDK_ROOT)/libc/xtensa-lx106-elf/lib -L$(SDK_ROOT)/lib -L$(SDK_ROOT)/ld -T$(FLASH_LAYOUT) -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy
+LD_STD_LIBS ?= -lhal -lphy -lnet80211 -llwip -lwpa -lmain -lpp -lsmartconfig -lwps -lcrypto -lespnow -laxtls -lc -lm -lgcc
 # stdc++ used in later versions of esp8266 Arduino
 LD_STD_CPP = lstdc++
 ifneq ($(shell grep $(LD_STD_CPP) $(ESP_ROOT)/platform.txt),)
-- 
2.8.1

```

Two linker flags, `-u _printf_float -u _scanf_float` can be made configurable via some `ENABLE_PRINTF_FLOAT`/`ENABLE_SCANF_FLOAT` options. Removing these options can save some memory for applications which don't use floating point formats.

plerup, 2016-07-28:
Time to create a parser of platform.txt, stay tuned...

igrr, 2016-07-28:
As another option, maybe we can solve this from the other end?
I can create a makefile with the right options and place it into the ESP8266 Arduino package. Then you can include the makefile and get all the options from there. This may be useful for other environments (platform.io?) as well.

plerup, 2016-07-28:
Well if you can live with updating both that file and platform.txt it would be very convenient.

The only thing you have to provide in this file is something like the current variables for the include, compile and link options

plerup, 2016-08-07:
I have a parsing solution in the works now. Has the advantage of functioning also with older versions.

plerup, 2016-08-07:
Now commited

igrr, 2016-08-09:
Wow, most impressive usage of perl inside a Makefile 👍 
Good point about older versions as well.

closed 2016-08-27 by plerup