plerup/makeEspArduino #14: Automatic detection of library include path fails for Servo library

igrr, 2016-07-28:
Servo library has header file in src/Servo.h, while source files are in src/esp8266/.
Include path which gets added is libraries/Servo/src/esp8266/, which causes `#include <Servo.h>` to fail.
Maybe path detection can be split into two parts: detect build directory separately from include directory, and use "*.h" find pattern when looking for include directory.

igrr, 2016-07-28:
This seems to work for me:
```diff
From 251ce75bdb87d0c66b929d1310a3d6b9a1ec6480 Mon Sep 17 00:00:00 2001
From: Ivan Grokhotkov <ivan@espressif.com>
Date: Thu, 28 Jul 2016 20:10:34 +0800
Subject: [PATCH] Fix library include path detection

---
 makeEspArduino.mk | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/makeEspArduino.mk b/makeEspArduino.mk
index aa9a03b..5f6891f 100644
--- a/makeEspArduino.mk
+++ b/makeEspArduino.mk
@@ -88,7 +88,7 @@ HTTP_TOOL = curl
 
 INCLUDE_DIRS += $(SDK_ROOT)/include $(SDK_ROOT)/lwip/include $(SDK_ROOT)/libc/xtensa-lx106-elf/include $(CORE_DIR) $(ESP_ROOT)/variants/generic $(OBJ_DIR)
 C_DEFINES = -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266
-C_INCLUDES = $(foreach dir,$(INCLUDE_DIRS) $(USER_DIRS),-I$(dir))
+C_INCLUDES = $(foreach dir,$(INCLUDE_DIRS) $(USER_INC_DIRS),-I$(dir))
 C_FLAGS ?= -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections
 CPP_FLAGS ?= -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections
 S_FLAGS ?= -c -g -x assembler-with-cpp -MMD
@@ -107,10 +107,12 @@ CORE_OBJ = $(patsubst %,$(OBJ_DIR)/%$(OBJ_EXT),$(notdir $(CORE_SRC)))
 CORE_LIB = $(OBJ_DIR)/core.ar
 
 # User defined compilation units
+USER_INC = $(SKETCH) $(shell find $(LIBS) -name "*.h")
 USER_SRC = $(SKETCH) $(shell find $(LIBS) -name "*.S" -o -name "*.c" -o -name "*.cpp")
 # Object file suffix seems to be significant for the linker...
 USER_OBJ = $(subst .ino,.cpp,$(patsubst %,$(OBJ_DIR)/%$(OBJ_EXT),$(notdir $(USER_SRC))))
 USER_DIRS = $(sort $(dir $(USER_SRC)))
+USER_INC_DIRS = $(sort $(dir $(USER_INC)))
 
 VPATH += $(shell find $(CORE_DIR) -type d) $(USER_DIRS)
 
-- 
2.8.1
```

plerup, 2016-07-28:
My intended solution for this kind of problem was that you add possible special include directories by defining them in INCLUDE_DIRS. That is the reason for the "+=" 

Your solution is of course more convenient however. Please do a pull request.

PR #15 (2016-07-28): Improve library include path detection
This change fixes #14 by looking specifically for library subfolder which contains .h files, as some libraries (e.g. Servo for esp8266) have a header file at top level and .cpp files in subdirectories.
commit 96cd1bb3 (2016-07-28): Improve library include path detection (#15)

This change fixes #14 by looking specifically for library subfolder which contains .h files, as some libraries (e.g. Servo for esp8266) have a header file at top level and .cpp files in subdirectories.
closed 2016-07-28 by plerup