plerup/makeEspArduino #20: bits/c++config.h problem

patrickelectric, 2016-10-28:
I got this error with the last version of this project and the [esp8266](https://github.com/esp8266/arduino).

In file included from /home/patrick/git/ibex/tomadinhaFeliz/Arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/tr1/cstdarg:32:0,
                 from /home/patrick/git/ibex/tomadinhaFeliz/Arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/tr1/stdarg.h:32,
                 from Arduino/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:39,
                 from Arduino/cores/esp8266/umm_malloc/umm_malloc.c:494:
/home/patrick/git/ibex/tomadinhaFeliz/Arduino/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/cstdarg:41:28: fatal error: bits/c++config.h: No such file or directory
 #include <bits/c++config.h>
                            ^
compilation terminated.
make: *** [Makefile:194: /tmp/mkESP/blink_nodemcu/umm_malloc.c.o] Error 1

`blink.ino

/*
 ESP8266 Blink by Simon Peter
 Blink the blue LED on the ESP-01 module
 This example code is in the public domain
 The blue LED on the ESP-01 module is connected to GPIO1
 (which is also the TXD pin; so we cannot use Serial.print() at the same time)
 Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
                                    // but actually the LED is on; this is because 
                                    // it is acive low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);                      // Wait for two seconds (to demonstrate the active low LED)
}
`

`Makefile
#====================================================================================
# makeESPArduino
#
# A makefile for ESP8286 Arduino projects.
# Edit the contents of this file to suit your project
# or just include it and override the applicable macros.
#
# License: GPL 2.1
# General and full license information is available at:
#    https://github.com/plerup/makeEspArduino
#
# Copyright (c) 2016 Peter Lerup. All rights reserved.
#
#====================================================================================

#====================================================================================
# Project specfic values
#====================================================================================

# Include possible project makefile. This can be used to override the defaults below
-include $(firstword $(PROJ_CONF) $(dir $(SKETCH))config.mk)

#=== Default values

# Main source file (sketch).
# If this variable is not specified the first sketch in current directory will be used.
# If none is found there, a demo example will be used instead.
SKETCH ?= ./blink.ino

# Includes in the sketch file of libraries from within the ESP8266 Arduino directories can be automatically
# detected but if this is not enough, define this variable with all libraries or directories needed.
[... 299 lines cut ...]
print "SIZE_COM=$$v{'recipe.size.pattern'}\n";
my $$flash_size = sprintf("0x%X", hex($$v{'build.spiffs_end'})-hex($$v{'build.spiffs_start'}));
print "MKSPIFFS_COM=$$v{'tools.mkspiffs.path'}/$$v{'tools.mkspiffs.cmd'} -b $$v{'build.spiffs_blocksize'} -s $$flash_size -c \$$(FS_DIR) \$$(FS_IMAGE)\n";
print "UPLOAD_COM=$$v{'upload.pattern'}\n";
my $$fs_upload_com = $$v{'upload.pattern'};
$$fs_upload_com =~ s/(.+ -ca) .+/$$1 $$v{'build.spiffs_start'} -cf \$$(FS_IMAGE)/;
print "FS_UPLOAD_COM=$$fs_upload_com\n";
my $$val = $$v{'recipe.hooks.core.prebuild.1.pattern'};
$$val =~ s/bash -c "(.+)"/$$1/;
$$val =~ s/(#define .+0x)(\`)/"\\$$1\"$$2/;
$$val =~ s/(\\)//;
print "PREBUILD_COM=$$val\n";
endef
export PARSE_ARDUINO

# Convert memory information
define MEM_USAGE
while (<>) {
  $$r += $$1 if /^\.(?:data|rodata|bss)\s+(\d+)/;
  $$f += $$1 if /^\.(?:irom0\.text|text|data|rodata)\s+(\d+)/;
}
print "\nMemory usage\n";
print sprintf("  %-6s %6d bytes\n" x 2 ."\n", "Ram:", $$r, "Flash:", $$f);
endef
export MEM_USAGE`

plerup, 2016-10-31:
What is your setup, OS and esp-Arduino installation?

Similar error can be found here: https://github.com/esp8266/Arduino/issues/1631

zakx, 2017-02-23:
I just had the same issue under OSX, and fixed it by moving the esp8266 git checkout out of my code hierarchy.

closed 2017-03-26 by plerup

rgkirch, 2017-10-22:
Hey. I had the same problem and I can recreate it. It doesn't work when the sketch is in the testDir folder but if I make a testDir/src folder and move the sketch into the src folder then it works.

    mkdir testDir
    cd testDir
    
    echo "cloning"
    git clone https://github.com/esp8266/Arduino.git esp8266
    git clone https://github.com/plerup/makeEspArduino.git
    git clone https://github.com/plerup/espsoftwareserial.git
    cd esp8266/tools
    
    echo "python esp8266/tools/get.py"
    python get.py
    cd ../..
    cp espsoftwareserial/examples/swsertest/swsertest.ino ./
    
    echo "building"
    make -f makeEspArduino/makeEspArduino.mk flash ESP_ROOT=/home/richie/testDir/esp8266 BOARD=nodemcuv2 SKETCH=/home/richie/testDir/src/swsertest.ino LIBS=/home/richie/testDir/espsoftwareserial/ > error.txt 2>&1
    mv error.txt ../

davidbanham, 2018-03-04:
I am seeing the same issue on Arch Linux. Moving the sketch file to a `src` directory gets it building correctly.