plerup/makeEspArduino #23: Link error when using config.mk or including makeEspArduino.mk

fitzterra, 2016-11-10:
When using the `makeEspArduino.mk` with all defaults i.e. `make -f ../../makeEspArduino/makeEspArduino.mk`, the sketch builds and links fine. When wanting to do per sketch build configuration by using either a `config.mk` file in my local sketch dir, or creating a `Makefile` which includes `makeEspArduino.mk`, I get a linker error:

```
Creating core archive
Linking /home/tomc/ownCloud/Electronics/ESP8266/sketchbook/tempMonitor/.build/tempMonitor_generic/tempMonitor.bin
  Versions: Unknown, 2.3.0
/home/tomc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file -Wl,--gc-sections: No such file or directory
collect2: error: ld returned 1 exit status
```

My environment is Arduino v1.6.11 on Linux with esp8266 v2.3.0 installed via the boards manager. The `makeEspArduino` finds all dependencies from the Arduino environment just fine.

My ESP8266 dev dir structure is as follows:
```
.
├── makeEspArduino
│   ├── LICENSE
│   ├── makeEspArduino.mk
│   └── README.md
└── sketchbook
    └── tempMonitor
        ├── config.mk
        └── tempMonitor.ino
```
The contents of `sketchbook/tempMonitor/config.mk` is:
```
# config.mk
THIS_DIR := $(realpath $(dir $(realpath $(lastword $(MAKEFILE_LIST)))))
ROOT := $(THIS_DIR)/..
#LIBS = $(ESP_LIBS)/SPI \
#  $(ESP_LIBS)/Wire \
#  $(ESP_LIBS)/ESP8266WiFi \
#  $(ROOT)/libraries \
#  $(ROOT)/ext_lib

FLASH_DEF = 512K64K
BUILD_DIR = $(THIS_DIR)/.build/$(MAIN_NAME)_$(BOARD)
```
and the `tempMonitor.ino` is a very simple sketch that reads the ADC and writes to the serial port, no special libs and it compiles and links fine as described above without the `config.mk` file.

I get the error mentioned above when running: ` make -f ../../makeEspArduino/makeEspArduino.mk` from within the `sketchbook/tempMonitor` dir.

Any help would be much appreciated in helping me understand what I am missing to do in `config.mk` for the linker to find the required parts.

Thanks,
 Tom

PS: I also tried setting up a `Makefile` which then includes `{correct_path}/makeEspArduino.mk`, but the error is the same, so the problem seems to be me messing up or not setting some var in my own make files.

plerup, 2016-11-10:
Change to
FLASH_DEF = 512k64

fitzterra, 2016-11-12:
That was the issue. Thanks very much!

closed 2016-11-12 by fitzterra