plerup/makeEspArduino #24: How to set output directory relative to the sketch

kommancs96, 2016-11-16:
I made an alias for the `makeEspArduino.mk` file as described in the readme. Now i want to change the makefile so that it produces a binary in the `/buid` folder next to the sketch. So if I have the sketch at `~/Documents/sketches/mySketch/fiest.ino` , the output should be placed at `~/Documents/sketches/mySketch/build/first.bin`

plerup, 2016-11-21:
Use:

**BUILD_DIR=./build**

Either in a config.mk or on the command line.

Just make sure to use the latest version though, 3.2.2

kommancs96, 2016-12-08:
Hello  I tried to modify the makefile so: `BUILD_DIR ?= $(dir $(shell pwd)/build)` , but this did not work.

plerup, 2016-12-08:
Ok, what happened then?

kommancs96, 2016-12-08:
It compiles in the same directory where the sketch is.

plerup, 2016-12-11:
Should be:

BUILD_DIR ?= $(dir $(shell pwd)/build/)
BUILD_DIR ?= $(shell pwd)/build

or preferably the solution I suggested in my first comment which doesn't require a change in the original makefile

kommancs96, 2016-12-11:
This worked, I close this issue.

closed 2016-12-11 by kommancs96