plerup/makeEspArduino #25: Can't compile .cpp and .c sketches

kommancs96, 2016-12-13:
I have the following simple sketch:
```

//Pin declarations
static const uint8_t D0   = 16;
static const uint8_t D1   = 5;
static const uint8_t D2   = 4;
static const uint8_t D3   = 0;
static const uint8_t D4   = 2;
static const uint8_t D5   = 14;
static const uint8_t D6   = 12;
static const uint8_t D7   = 13;
static const uint8_t D8   = 15;
static const uint8_t D9   = 3;
static const uint8_t D10  = 1;

extern "C" {
#include "user_interface.h"
#include "ets_sys.h"
#include "osapi.h"
}
static uint8_t value = 0;
static uint8_t lastValue = 0;

void setup(){
    Serial.begin(115200);
    Serial.printf("Initializing pins ...\n");
    pinMode(D3, OUTPUT);
    pinMode(D5, OUTPUT);
    pinMode(D6, OUTPUT);
}
void loop() {
    lastValue = analogRead(A0);
    if (value > lastValue) {
        digitalWrite(D5, 1);
        delay(40);
        digitalWrite(D5, 0);
    }else if(value < lastValue){
        digitalWrite(D6, 1);
        delay(40);
        digitalWrite(D6, 0);
    }else if(value == lastValue){
        digitalWrite(D3, 1);
        delay(40);
        digitalWrite(D3, 0);
    }
    value = lastValue;
    Serial.println(value);
    delay(100);
}

```
If I save this sketch as `analog.ino` and then run `espmake SKETCH=analog.ino` then it compiles sucessfully. But if I chnge the extension to `.cpp` and then run `espmake SKETCH=analog.cpp`, I get a lot of errors saying Serial is not declared, A0 is not declared like these:
```
analog.cpp:38:3: error: 'Serial' undeclared (first use in this function)
   Serial.begin(115200);
analog.cpp:14:1: error: unknown type name 'uint8_t'
 static const uint8_t D0   = 16;
```

plerup, 2016-12-13:
It is only .ino files who gets an implicit include of Arduino.h  This is for Arduino IDE legacy reasons.

If you have a c or cpp file you have to add:

#include <Arduino.h>

kommancs96, 2016-12-13:
@plerup It's manually added when I comile .cpp or .c The errors are not compile-time error, but linker errors.

plerup, 2016-12-14:
Tried your example now (after adding #include <stdint.h>) and it builds without any problem.

The fact that you for some reason get CMakeFiles/3.6.2 in your link operations indicates something strange in your setup.

kommancs96, 2016-12-15:
CMakeFiles is not strange, since I try to configure CLion for esp8266 development, which uses CMake.
Added `stdint.h`, but it did not solve my problem. Here are the errors I get:

```
Linking /path/to/project/build/ADCThermo.bin
  Versions: Unknown, 2.3.0
/path/to/project/build/CMakeCXXCompilerId.cpp.o:(.data.info_compiler+0x0): multiple definition of `info_compiler'
/path/to/project/build/CMakeCCompilerId.c.o:(.data.info_compiler+0x0): first defined here
/path/to/project/build/CMakeCXXCompilerId.cpp.o:(.data.info_platform+0x0): multiple definition of `info_platform'
/path/to/project/build/CMakeCCompilerId.c.o:(.data.info_platform+0x0): first defined here
/path/to/project/build/CMakeCXXCompilerId.cpp.o:(.data.info_language_dialect_default+0x0): multiple definition of `info_language_dialect_default'
/path/to/project/build/CMakeCCompilerId.c.o:(.data.info_language_dialect_default+0x0): first defined here
/path/to/project/build/CMakeCXXCompilerId.cpp.o: In function `main':
/path/to/project//cmake-build-debug/CMakeFiles/3.6.2/CompilerIdCXX/CMakeCXXCompilerId.cpp:516: multiple definition of `main'
/path/to/project/build/CMakeCCompilerId.c.o:/path/to/project//cmake-build-debug/CMakeFiles/3.6.2/CompilerIdC/CMakeCCompilerId.c:542: first defined here
/path/to/project/build/CMakeCXXCompilerId.cpp.o:(.data.info_arch+0x0): multiple definition of `info_arch'
/path/to/project/build/CMakeCCompilerId.c.o:(.data.info_arch+0x0): first defined here
/path/to/project/build/feature_tests.c.o: In function `main':
/path/to/project//cmake-build-debug/CMakeFiles/feature_tests.c:34: multiple definition of `main'
/path/to/project/build/CMakeCCompilerId.c.o:/path/to/project//cmake-build-debug/CMakeFiles/3.6.2/CompilerIdC/CMakeCCompilerId.c:542: first defined here
collect2: error: ld returned 1 exit status
make[4]: *** [/path/to/project/build/ADCThermo.bin] Error 1
make[3]: *** [CMakeFiles/test_makefile] Error 2
make[2]: *** [CMakeFiles/test_makefile.dir/all] Error 2
make[1]: *** [CMakeFiles/test_makefile.dir/rule] Error 2
make: *** [test_makefile] Error 2
```

plerup, 2016-12-16:
Use

VERBOSE=1

in order to show what's actually happening here

closed 2017-01-29 by plerup