plerup/makeEspArduino #26: Wrong pins_arduino.h used

brent-blood, 2017-01-22:
I cannot get builds from the makefile to use the pin definitions for the specific board that I'm using (Node MCU 1.0). As you'll see below, I specified BOARD in the makefile as 'nodemcu' which as far as I can tell is the correct value.

My simple sketch to test that I get the same behavior using the makefile as I get with the IDE:

```
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  Serial.begin(115200);
  delay(100);

  Serial.printf("LED_BUILTIN: %d", LED_BUILTIN);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(250);
  digitalWrite(LED_BUILTIN, LOW);
  delay(250);
}
```

My makefile looks like:

```
SKETCH = esp8266_blink.ino

UPLOAD_PORT = /dev/tty.wchusbserial1420
BOARD = nodemcu

include $(HOME)/makeEspArduino/makeEspArduino.mk
```

When I build and upload that using the Arduino IDE the builtin LED blinks and the serial monitor tells me at LED_BUILTIN is pin 16. When I build and upload using the makefile, serial output tells me that it's using pin 1 (I assume from the default esp8266 version of pins_arduino.h rather than nodemcu). If I change the sketch to refer to 16 explicitly for the pin, it works as expected, so this is just the wrong pin definition header being used.

plerup, 2017-01-23:
I will change this but currently you have to specify the variable INCLUDE_VARIANT as well

INCLUDE_VARIANT = nodemcu

brent-blood, 2017-01-23:
Thanks. Making a change to not require this extra definition would be the best move in my opinion, but you could also just call this out in README.md as part of an example. Along those same lines, I wasn't able to easily tell how to find the possible values (and thus the correct one for me) to use for BOARD so a note on that would help people get going as well (something like "list the contents of 'hardware/arduino/avr/variants/' within your Arduino IDE installation" - or whatever the recommended method is).

That said, great work and thanks for sharing it!

plerup, 2017-01-29:
This is now handled automatically in the latest relaease

closed 2017-02-23 by plerup