plerup/makeEspArduino #31: Linking against libmain.a

kommancs96, 2017-03-16:
How do I link against any specific `.a` file?
The reason why I'm asking this is because I have the following code:
```
#include "spi_flash.h"
void testSPIFlash() {
  uint32_t* buf = (uint32_t*)calloc(8, sizeof(uint32_t));
  uint32_t from_addr = 0x100000;
  uint32_t to_addr = from_addr + 0xfb000;

  for (uint32_t i = from_addr; i < to_addr; i+=8) {
    spi_flash_read(i, buf, 8*sizeof(uint32_t));
    Serial.print(i, HEX);
    Serial.print(":\t");
    for (size_t i = 0; i < 8; i++) {
      Serial.print(buf[i], HEX);
      Serial.print(" ");
    }
    Serial.println();
  }
}

void setup() {
 testSPIFlash();
}
void loop(){
}
```
The code compiles, but I get a linker error:
`undefined reference to spi_flash_read(unsigned int, unsigned int*, unsigned int)`
After demangling symbols in `/esp8266_arduino/esp8266/tools/sdk/lib/*.a` I found that `libmain.a` cointains a function named `spi_flash_read`. So it seems like the makefile does not link against `libmain.a`.

plerup, 2017-03-26:
Well the libraries used is controlled by the Arduino config file platform.tx and there the "main" library is referenced. Please check your file, the following line should be there

compiler.c.elf.libs=-lhal -lphy -lpp -lnet80211 {build.lwip_lib} -lwpa -lcrypto _**-lmain**_ -lwps -laxtls -lespnow -lsmartconfig -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc

zgoda, 2017-05-05:
Add:

```C++
extern "C" void spi_flash_read(uint32_t a, uint32_t * b, uint32_t c);
```

or whatever signature this function has (specially return value).

closed 2017-05-21 by plerup