plerup/makeEspArduino #32: xtensa-lx106-elf-g++: not found

rgkirch, 2017-03-25:
```
➜  git:(functional) ✗ make -f makeEspArduino.mk ESP_ROOT=~/Documents/github/esp8266/Arduino SKETCH=src/main.cpp   
debug.cpp
WMath.cpp
core_esp8266_wiring_analog.c
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: not found
HardwareSerial.cpp
makeEspArduino.mk:180: recipe for target '/tmp/mkESP/main_generic/debug.cpp.o' failed
make: *** [/tmp/mkESP/main_generic/debug.cpp.o] Error 127
make: *** Waiting for unfinished jobs....
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: not found
makeEspArduino.mk:180: recipe for target '/tmp/mkESP/main_generic/WMath.cpp.o' failed
make: *** [/tmp/mkESP/main_generic/WMath.cpp.o] Error 127
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found
makeEspArduino.mk:188: recipe for target '/tmp/mkESP/main_generic/core_esp8266_wiring_analog.c.o' failed
make: *** [/tmp/mkESP/main_generic/core_esp8266_wiring_analog.c.o] Error 127
core_esp8266_timer.c
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: not found
makeEspArduino.mk:180: recipe for target '/tmp/mkESP/main_generic/HardwareSerial.cpp.o' failed
make: *** [/tmp/mkESP/main_generic/HardwareSerial.cpp.o] Error 127
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found
makeEspArduino.mk:188: recipe for target '/tmp/mkESP/main_generic/core_esp8266_timer.c.o' failed
make: *** [/tmp/mkESP/main_generic/core_esp8266_timer.c.o] Error 127
➜  git:(functional) ✗ ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ --version
xtensa-lx106-elf-g++ (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I haven't spent any time debugging this. I see a "not found" and then it gives me the path to what it couldn't find. I set it up following the instructions in the readme and all of the steps seemed pretty clear.

I just tried it on a small cpp file to see what happens.
```
➜  ~ make -f ~/Documents/github/plerub/makeEspArduino/makeEspArduino.mk ESP_ROOT=~/Documents/github/esp8266/Arduino SKETCH=make-unique-test.cpp    

find: ‘./.cache/dconf’: Permission denied
find: ‘./.cache/dconf’: Permission denied
find: ‘./.cache/dconf’: Permission denied
find: ‘./.cache/dconf’: Permission denied
core_esp8266_wiring_analog.c
HardwareSerial.cpp
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: not found
/home/richie/Documents/github/plerub/makeEspArduino/makeEspArduino.mk:188: recipe for target '/tmp/mkESP/make-unique-test_generic/core_esp8266_wiring_analog.c.o' failed
make: *** [/tmp/mkESP/make-unique-test_generic/core_esp8266_wiring_analog.c.o] Error 127
make: *** Waiting for unfinished jobs....
/bin/sh: 1: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: not found
/home/richie/Documents/github/plerub/makeEspArduino/makeEspArduino.mk:180: recipe for target '/tmp/mkESP/make-unique-test_generic/HardwareSerial.cpp.o' failed
make: *** [/tmp/mkESP/make-unique-test_generic/HardwareSerial.cpp.o] Error 127
```
The file that I was trying to compile was just this.
```
#include <memory>
#include <string>

template <typename T>
class Nom
{
public:
    Nom(T d) : data(d) {};
    T data;
};

int main()
{
    auto value = std::make_unique<Nom<int>>(6);
    return  #0;
}
```
I have been using platformio but I suddenly want to use std::make_unique and I haven't figured out how to build with c++14. I'll want to upload this code to an esp8266. I'm running Ubuntu 16.04.
```
➜  ~ uname -a
Linux chuwi 4.8.0-41-generic #44~16.04.1-Ubuntu SMP Fri Mar 3 17:11:16 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```
Thanks for the work! I hope I can figure out how to build with c++14.

plerup, 2017-03-25:
Looks like the compiler isn't there. Please verify: ~/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc

Did the "python get.py" execute ok?

rgkirch, 2017-03-27:
```
➜  tools git:(master) pwd
/home/richie/Documents/github/esp8266/Arduino/tools
➜  tools git:(master) python get.py 
Platform: x86_64-pc-linux-gnu
Tool esptool-0.4.9-linux64.tar.gz already downloaded
Extracting dist/esptool-0.4.9-linux64.tar.gz
Renaming esptool-0.4.9-linux64 to esptool
Tool linux64-xtensa-lx106-elf-gb404fb9.tar.gz already downloaded
Extracting dist/linux64-xtensa-lx106-elf-gb404fb9.tar.gz
Tool mkspiffs-0.1.2-linux64.tar.gz already downloaded
Extracting dist/mkspiffs-0.1.2-linux64.tar.gz
Renaming mkspiffs-0.1.2-linux64 to mkspiffs
```
I'm pretty sure it didn't have any errors the first time that I ran it.
```
➜  bin git:(master) pwd
/home/richie/Documents/github/esp8266/Arduino/tools/xtensa-lx106-elf/bin
➜  bin git:(master) ./xtensa-lx106-elf-gcc --version
xtensa-lx106-elf-gcc (crosstool-NG 1.20.0) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

➜  bin git:(master) ls
xtensa-lx106-elf-addr2line  xtensa-lx106-elf-ct-ng.config  xtensa-lx106-elf-gcc-ranlib  xtensa-lx106-elf-objcopy
xtensa-lx106-elf-ar         xtensa-lx106-elf-elfedit       xtensa-lx106-elf-gcov        xtensa-lx106-elf-objdump
xtensa-lx106-elf-as         xtensa-lx106-elf-g++           xtensa-lx106-elf-gdb         xtensa-lx106-elf-ranlib
xtensa-lx106-elf-c++        xtensa-lx106-elf-gcc           xtensa-lx106-elf-gprof       xtensa-lx106-elf-readelf
xtensa-lx106-elf-cc         xtensa-lx106-elf-gcc-4.8.2     xtensa-lx106-elf-ld          xtensa-lx106-elf-size
xtensa-lx106-elf-c++filt    xtensa-lx106-elf-gcc-ar        xtensa-lx106-elf-ld.bfd      xtensa-lx106-elf-strings
xtensa-lx106-elf-cpp        xtensa-lx106-elf-gcc-nm        xtensa-lx106-elf-nm          xtensa-lx106-elf-strip
```
It looks like the right files are in the right places...

plerup, 2017-03-27:
Try changing:
ESP_ROOT=~/Documents/github/esp8266/Arduino
to:
ESP_ROOT=/home/richie/Documents/github/esp8266/Arduino

rgkirch, 2017-03-28:
Hey, I just wanted to let you know that I tried using an absolute path this morning and it worked. That is the solution, thanks! I'm still fooling around trying to get a project to build though. I'll let you know when I get a project working or if I can't figure it out.

closed 2017-04-12 by plerup