plerup/makeEspArduino #35: Extra -o on Windows causes no files to compile

jontio, 2017-04-25:
Running on Windows it fails with "xtensa-lx106-elf-g++.exe: fatal error: no input files"

looking more closely with GNU Make 4.2.1 I see output like...

```
/e/git-junk/hardware/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/e/git-junk/hardware/esp8266/tools/sdk/include -I/e/git-junk/hardware/esp8266/tools/sdk/lwip/include -I/e/git-junk/hardware/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -IE:/msys64/tmp/mkESP/test_generic/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01" -DESP8266 -I/e/git-junk/hardware/esp8266/cores/esp8266 -I/e/git-junk/hardware/esp8266/variants/generic -IE:/msys64/tmp/mkESP/test_generic -IE:/git-junk/makeEspArduino/arduino/ -o /e/git-junk/hardware/esp8266/cores/esp8266/abi.cpp -o E:/msys64/tmp/mkESP/test_generic/abi.cpp.o
```
There is an extra -o before the cpp file.

This comes from the of CPP_COM that is the following for me

```
/e/git-junk/hardware/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/e/git-junk/hardware/esp8266/tools/sdk/include -I/e/git-junk/hardware/esp8266/tools/sdk/lwip/include -I/e/git-junk/hardware/esp8266/tools/sdk/libc/xtensa-lx106-elf/include -IE:/msys64/tmp/mkESP/test_generic/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD="ESP8266_ESP01" -DESP8266 -I/e/git-junk/hardware/esp8266/cores/esp8266 -I/e/git-junk/hardware/esp8266/variants/generic -IE:/msys64/tmp/mkESP/test_generic -IE:/git-junk/makeEspArduino/arduino/ -o
```

The only place I see CPP_COM set in ```print "CPP_COM=$$v{'recipe.cpp.o.pattern'}\n";``` but as I don't understand makefiles well enough I'm not sure how it is being set.

How can I remove this pesky -o ?

Maybe a sanity check for CPP_COM that if it sees a -o at the end it removes it.

jontio, 2017-04-25:
Putting in 
```
TMPVAR := $(CPP_COM)
CPP_COM = $(filter-out -o, $(TMPVAR))
```
at line 175 removes the trailing -o

But now make ends with...

```
xtensa-lx106-elf-gcc.exe: warning: '-x assembler-with-cpp' after last input file has no effect
xtensa-lx106-elf-gcc.exe: fatal error: no input files
compilation terminated.
makeEspArduino.mk:195: recipe for target 'E:/msys64/tmp/mkESP/HelloServer_generic/cont.S.o' failed
mingw32-make: *** [E:/msys64/tmp/mkESP/HelloServer_generic/cont.S.o] Error 1
mingw32-make: *** Waiting for unfinished jobs....
abi.cpp
base64.cpp
cbuf.cpp
```

jontio, 2017-04-25:
It seems the same was happening to S_COM and C_COM. Putting in the following at line 175 removes all the pesky -os. Now the demo application compiles. 

```
TMPVAR := $(CPP_COM)
CPP_COM = $(filter-out -o, $(TMPVAR))

TMPVAR2 := $(S_COM)
S_COM = $(filter-out -o, $(TMPVAR2))

TMPVAR3 := $(C_COM)
C_COM = $(filter-out -o, $(TMPVAR3))
```

I'm not sure how the C_COM S_COM CC_COM variables are generated and why there are trailing -os. It's a rather nasty hack but it's my best attempt at getting this makefile to work so far.

plerup, 2017-04-25:
What Linux subsystem for Windows are you using? Only cygwin is supported (tested)

jontio, 2017-04-25:
I use MSYS2 and pacman to install packages. I have been using MinGW along with `mingw32-make.exe` that says
```
GNU Make 4.1.90
Built for Windows32
```
I can run make from a standard windows shell but prefer the MSYS2, so I assume that I am using only native window applications.

I notice you are also using perl, my version of that says `This is perl 5, version 24, subversion 1 (v5.24.1) built for x86_64-msys-thread-multi`

But no, I don't have cygwin. However adding the following code to line 175 makes your makefile work fine as far as I have tested so far.

```
TMPVAR := $(CPP_COM)
CPP_COM = $(filter-out -o, $(TMPVAR))
TMPVAR2 := $(S_COM)
S_COM = $(filter-out -o, $(TMPVAR2))
TMPVAR3 := $(C_COM)
C_COM = $(filter-out -o, $(TMPVAR3))
```

My desire to use your program was so I could write a wizard template for Qt Creator and integrate esp8266 programming from Qt Creator. I have done that https://github.com/jontio/qt_esp8266_wizard and have had no troubles with it so far.

plerup, 2017-04-26:
It is really strange as the trailing -o is to be removed by this line in the Perl code

    $$v{$$key} =~ s/ -o $$//;

Try changing it to

    $$v{$$key} =~ s/ -o\s+$$//;

jontio, 2017-04-26:
I tried that and it works :)

So I think that means on my setup I have two spaces after the -o if I understand regular expressions in Perl correctly.

Thanks heaps

plerup, 2017-04-26:
I would rather guess that it's the handling of line breaks that may differ in the Perl you are running.

I will add this change in the next commit. Thanks for reporting.

plerup, 2017-05-22:
Included in latest commit

jontio, 2017-05-23:
Thanks. It works fine without any modification now.

closed 2017-05-23 by jontio