plerup/makeEspArduino #37: No more "foo as not declared in this scope"

nemo9955, 2017-06-01:
Basically in Arduino IDE you can call a function and declare it AFTER the first one.
This make file does not do that, witch I think causes an error " 'STAILQ_NEXT' was not declared in this scope" when I flash [PainlessMesh](https://gitlab.com/BlackEdder/painlessMesh). The exact same sketch and lib work with the Arduino IDE.

This is the sketch : 
https://gitlab.com/BlackEdder/painlessMesh/blob/master/examples/startHere/startHere.ino
NOTE: using the .ino above with the makefile I also get 
`'receivedCallback' was not declared in this scope
'newConnectionCallback' was not declared in this scope
etc.` witch are solved by simply moving the `setup` and `loop` functions at the end of the file (error does not apear in Arduino IDE).

I think there is also a problem in the "automatic" library detection, note how I have to explicitly include 

Command and output :
```
make -f /PATH1/deploy-libs/makeEspArduino/makeEspArduino.mk ESP_ROOT=/PATH1/deploy-libs/esp8266 
LIBS="/PATH1/deploy-libs/esp8266/libraries /PATH1/sketch-libs "  BOARD=d1_mini   -L  
flash  SKETCH=/PATH2/mesh/mesh.ino

* Build state has changed, doing a full rebuild *
core_esp8266_phy.c
abi.cpp
base64.cpp
cbuf.cpp
cont.S
cont_util.c
core_esp8266_eboot_command.c
core_esp8266_flash_utils.c
core_esp8266_i2s.c
core_esp8266_noniso.c
core_esp8266_main.cpp
pgmspace.cpp
Schedule.cpp
Print.cpp
spiffs_cache.c
spiffs_gc.c
spiffs_nucleus.c
spiffs_check.c
spiffs_hydrogen.c
spiffs_api.cpp
spiffs_hal.cpp
StreamString.cpp
Stream.cpp
time.c
Tone.cpp
uart.c
umm_malloc.c
Updater.cpp
core_esp8266_si2c.c
core_esp8266_postmortem.c
core_esp8266_timer.c
core_esp8266_wiring.c
WMath.cpp
WString.cpp
core_esp8266_wiring_analog.c
core_esp8266_wiring_digital.c
core_esp8266_wiring_pulse.c
core_esp8266_wiring_shift.c
core_esp8266_wiring_pwm.c
FS.cpp
debug.cpp
heap.c
HardwareSerial.cpp
Esp.cpp
IPAddress.cpp
cencode.c
cdecode.c
libc_replacements.c
MD5Builder.cpp
mesh.ino
ArduinoOTA.cpp
DNSServer.cpp
EEPROM.cpp
ESP8266AVRISP.cpp
ESP8266HTTPClient.cpp
ESP8266HTTPUpdateServer.cpp
ESP8266httpUpdate.cpp
ESP8266LLMNR.cpp
ESP8266mDNS.cpp
ESP8266NetBIOS.cpp
ESP8266SSDP.cpp
ESP8266WebServer.cpp
Parsing.cpp
ESP8266WiFi.cpp
ESP8266WiFiAP.cpp
ESP8266WiFiGeneric.cpp
ESP8266WiFiMulti.cpp
ESP8266WiFiScan.cpp
ESP8266WiFiSTA.cpp
WiFiClient.cpp
WiFiClientSecure.cpp
WiFiServer.cpp
WiFiUdp.cpp
ESP8266WiFiMesh.cpp
Dns.cpp
Dhcp.cpp
Ethernet.cpp
EthernetClient.cpp
EthernetServer.cpp
EthernetUdp.cpp
socket.cpp
w5100.cpp
gdbstub-entry.S
gdbstub.c
Hash.cpp
sha1.c
File.cpp
SD.cpp
Sd2Card.cpp
SdFile.cpp
SdVolume.cpp
Servo.cpp
SPI.cpp
hspi_slave.c
SPISlave.cpp
font.c
TFTv2.cpp
Ticker.cpp
Adafruit_BMP085.cpp
DHT.cpp
Wire.cpp
DHT_U.cpp
painlessMesh.cpp
painlessMeshAP.cpp
painlessMeshComm.cpp
painlessMeshConnection.cpp
painlessMeshDebug.cpp
painlessMeshSync.cpp
painlessMeshSTA.cpp
/PATH1/sketch-libs/PainlessMesh/src/painlessMeshSTA.cpp: In member function 'void StationScan::scanComplete(bss_info*)':
/PATH1/sketch-libs/PainlessMesh/src/painlessMeshSTA.cpp:185:44: error: 'STAILQ_NEXT' was not declared in this scope
         bssInfo = STAILQ_NEXT(bssInfo, next);
                                            ^
Creating core archive
make: *** [/tmp/mkESP/mesh_d1_mini/painlessMeshSTA.cpp.o] Error 1
/PATH1/deploy-libs/makeEspArduino/makeEspArduino.mk:198: recipe for target '/tmp/mkESP/mesh_d1_mini/painlessMeshSTA.cpp.o' failed
make: *** Waiting for unfinished jobs....

```

plerup, 2017-06-01:
This obscure function of the Arduino IDE to automatically add forward declarations of all functions is not standard C/C++ and hence should be avoided I think. It would require automatic patching of the main sketch file and that is not going to be the case in this makefile.

nemo9955, 2017-06-02:
I understand that, but still, `mesh.ino` builds before all the PM libs, witch may be the cause of the problem.
Shouldn't the sketch be the last one to build ?

plerup, 2017-06-02:
Build order never matters for c-files as long as all h-files are static.

closed 2017-07-07 by plerup