plerup/makeEspArduino #39: esp32: path to esptool is incorrect

briancappello, 2017-06-22:
I think I've figured out what's going on, but I'm not quite sure how to fix it (having no perl skills..)

`ELF2BIN_COM` gets set from `platform.txt`'s `recipe.objcopy.hex.pattern`, which starts with `{tools.esptool.cmd}` - however that needs to be substituted for `{tools.esptool.cmd.windows}` if on windows, otherwise `{tools.esptool.cmd.linux}`. The platform-specific keys in `platform.txt` have the correct full paths to `esptool.py` or `esptool.exe`. 

As a workaround I just symlinked `esptool -> esptool.py` in my `ESP_ROOT/tools` directory, and that does work, but I don't think its a proper solution...

everslick, 2017-07-04:
I have the same issue - also building under Linux for ESP32.

plerup, 2017-07-07:
Fixed in latest version

everslick, 2017-07-07:
I can confirm, that it works now! Thank you very much!

briancappello, 2017-07-07:
Works for me too, thanks!

closed 2017-07-07 by briancappello