plerup/makeEspArduino #41: reliability

teliot, 2017-06-30:
With the arduino IDE the esp12 give me no trouble. with this make file the esp will WDT reset dozens of times before it gets a successful run. even with a basic blink sketch.

plerup, 2017-07-07:
Strange, I build for esp12 without problems.

Please describe your setup

teliot, 2017-07-08:
same setup for makefile and arduino.

ESP12 module power from a breadboard power supply (AMS1117) which is powered from a XBOX PSU (12v switch mode with voltage sense to maintain 12v at all loads)

the board has a ZS-042 module (DS3231 RTC)

Both my clone prolific YP-01 and my authentic SIL2104 work fine in arduino but give me probs with the makefile.

my make project in questions in teliot/DS3231
it's makefile is (typed, not pasted)
LIBS = /home/......./project directory   \
          /home/...../esp8266/libraries/Wire
SKETCH = DS3231.ino
include /home/....../myMakefile.mk

myMakefile.mk contains
LIBS += /home/...../esp8266/cores/esp8266/Arduino.h
ESP_ROOT = /home/user/esp8266  (this folder is the github/esp8266/Arduino repo)
UPLOAD_SPEED=115200
UPLOAD_PORT=/dev/ttyUSB0
BOARD=d1
include /home/...../makeEspArduino/makeEspArduino.mk

plerup, 2017-07-11:
Try using the "list_lib" action to see which files that are used.

You can also use the VERBOSE=1 option to get a complete listing on what the actual build operations are

everslick, 2017-11-12:
I think this issue can safely be closed, no?

closed 2017-11-12 by plerup