plerup/makeEspArduino #42: Make - no rule to make target

BarrettLowe, 2017-07-07:
I know this error is a very basic one but I can't figure it out...

Your makefile is above my expertise too. Any ideas on this one? I'm trying to include a third party library. The error I get is:
`make: *** No rule to make target `SFE_BMP180.cpp', needed by `/tmp/mkESP/LoweWeather_nodemcuv2/SFE_BMP180.cpp.o'.  Stop.`

I *think* the path to that specific cpp is in my LIBS variable. I'm on OSX developing with a NodeMCU esp8266. My call is:
`make -f ./makeEspArduino.mk SKETCH=./LoweWeather.ino BOARD=nodemcuv2`
I tried to ignore the BOARD but on simpler sketches, it wouldn't upload at all...

plerup, 2017-07-10:
Try the "list_lib" action to show the files which the makefile is looking for

BarrettLowe, 2017-07-21:
Did some futzing around but can't seem to get it. When I make with list_lib it says that I'm including the path to the h file and the cpp file is listed under source files. I did notice that at the beginning of my make call I also get the following line ...

`/bin/sh: /proc/94666/cmdline: No such file or directory`

plerup, 2017-07-21:
Please add VERBOSE=1 to the command line and then provide the complete printout

BarrettLowe, 2017-07-21:
Sure. Thanks for the help!

`barrettlowe in ~/Documents/Arduino/LoweWeather on master ● ● λ espmake SKETCH=LoweWeather.cpp BOARD=nodemcuv2 list_lib
* Build state has changed, doing a full rebuild *
=== User specific libraries ===
* Include directories:
.//
/Users/barrettlowe/esp8266/libraries//Adafruit-BMP085-Library/
/Users/barrettlowe/esp8266/libraries//Adafruit_INA219/
/Users/barrettlowe/esp8266/libraries//Adafruit_MLX90614_Library/
/Users/barrettlowe/esp8266/libraries//Adafruit_Unified_Sensor/
/Users/barrettlowe/esp8266/libraries//ArduinoOTA/
/Users/barrettlowe/esp8266/libraries//AsyncDelay/src/
/Users/barrettlowe/esp8266/libraries//BMP180_Breakout_Arduino_Library/src/
/Users/barrettlowe/esp8266/libraries//DHT_sensor_library/
/Users/barrettlowe/esp8266/libraries//DNSServer/src/
/Users/barrettlowe/esp8266/libraries//EDB/
/Users/barrettlowe/esp8266/libraries//EEPROM/
/Users/barrettlowe/esp8266/libraries//ESP8266AVRISP/src/
/Users/barrettlowe/esp8266/libraries//ESP8266HTTPClient/src/
/Users/barrettlowe/esp8266/libraries//ESP8266HTTPUpdateServer/src/
/Users/barrettlowe/esp8266/libraries//ESP8266LLMNR/
/Users/barrettlowe/esp8266/libraries//ESP8266NetBIOS/
/Users/barrettlowe/esp8266/libraries//ESP8266SSDP/
/Users/barrettlowe/esp8266/libraries//ESP8266WebServer/src/
/Users/barrettlowe/esp8266/libraries//ESP8266WebServer/src/detail/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/include/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFiMesh/src/
/Users/barrettlowe/esp8266/libraries//ESP8266httpUpdate/src/
/Users/barrettlowe/esp8266/libraries//ESP8266mDNS/
/Users/barrettlowe/esp8266/libraries//Ethernet/examples/udpntpclient_pedantic/
/Users/barrettlowe/esp8266/libraries//Ethernet/src/
/Users/barrettlowe/esp8266/libraries//Ethernet/src/utility/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/internal/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/xtensa/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/xtensa/config/
/Users/barrettlowe/esp8266/libraries//Hash/src/
/Users/barrettlowe/esp8266/libraries//Hash/src/sha1/
/Users/barrettlowe/esp8266/libraries//MemoryFree/
/Users/barrettlowe/esp8266/libraries//SD/src/
/Users/barrettlowe/esp8266/libraries//SD/src/utility/
/Users/barrettlowe/esp8266/libraries//SPI/
/Users/barrettlowe/esp8266/libraries//SPISlave/src/
/Users/barrettlowe/esp8266/libraries//Servo/src/
/Users/barrettlowe/esp8266/libraries//Servo/src/esp8266/
/Users/barrettlowe/esp8266/libraries//SoftWire/src/
/Users/barrettlowe/esp8266/libraries//TFT_Touch_Shield_V2/
/Users/barrettlowe/esp8266/libraries//Ticker/
/Users/barrettlowe/esp8266/libraries//Wire/
/Users/barrettlowe/esp8266/libraries//esp8266/src/
/Users/barrettlowe/esp8266/libraries/BMP180_Breakout_Arduino_Library//src/
/Users/barrettlowe/esp8266/libraries/ESP8266HTTPUpdateServer//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer//src/detail/
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi//src/include/
/Users/barrettlowe/esp8266/libraries/SoftWire//src/
* Library source files:
LoweWeather.cpp
.//ESP_httplib.cpp
.//LoweWeather.cpp
.//weatherCalcs.cpp
/Users/barrettlowe/esp8266/libraries/Adafruit_MLX90614_Library/Adafruit_MLX90614.cpp
/Users/barrettlowe/esp8266/libraries/BMP180_Breakout_Arduino_Library/src/SFE_BMP180.cpp
/Users/barrettlowe/esp8266/libraries/DHT_sensor_library/DHT.cpp
/Users/barrettlowe/esp8266/libraries/DHT_sensor_library/DHT_U.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer/src/Parsing.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiAP.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiScan.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/WiFiClientSecure.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/WiFiServer.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi/src/WiFiUdp.cpp
/Users/barrettlowe/esp8266/libraries/SoftWire/src/SoftWire.cpp
/Users/barrettlowe/esp8266/libraries/SPI/SPI.cpp

barrettlowe in ~/Documents/Arduino/LoweWeather on master ● ● λ espmake SKETCH=LoweWeather.cpp BOARD=nodemcuv2
/bin/sh: /proc/99049/cmdline: No such file or directory
* Build state has changed, doing a full rebuild *
abi.cpp
base64.cpp
cbuf.cpp
cont.S
cont_util.c
core_esp8266_eboot_command.c
core_esp8266_flash_utils.c
core_esp8266_i2s.c
core_esp8266_main.cpp
core_esp8266_noniso.c
core_esp8266_phy.c
core_esp8266_postmortem.c
core_esp8266_si2c.c
core_esp8266_timer.c
core_esp8266_wiring.c
core_esp8266_wiring_analog.c
core_esp8266_wiring_digital.c
core_esp8266_wiring_pulse.c
core_esp8266_wiring_pwm.c
core_esp8266_wiring_shift.c
debug.cpp
Esp.cpp
FS.cpp
HardwareSerial.cpp
heap.c
IPAddress.cpp
cdecode.c
cencode.c
libc_replacements.c
MD5Builder.cpp
pgmspace.cpp
Print.cpp
Schedule.cpp
spiffs_cache.c
spiffs_check.c
spiffs_gc.c
spiffs_hydrogen.c
spiffs_nucleus.c
spiffs_api.cpp
spiffs_hal.cpp
Stream.cpp
StreamString.cpp
time.c
Tone.cpp
uart.c
umm_malloc.c
Updater.cpp
WMath.cpp
WString.cpp
Creating core archive
LoweWeather.cpp
ESP_httplib.cpp
weatherCalcs.cpp
make: *** No rule to make target 'Adafruit_MLX90614.cpp', needed by '/tmp/mkESP/LoweWeather_nodemcuv2/Adafruit_MLX90614.cpp.o'.  Stop.

`

Thanks again

BarrettLowe, 2017-07-21:
Sure. Thanks for the help

barrettlowe in ~/Documents/Arduino/LoweWeather on master ● ● λ espmake SKETCH=LoweWeather.cpp BOARD=nodemcuv2 list_lib
* Build state has changed, doing a full rebuild *
=== User specific libraries ===
* Include directories:
.//
/Users/barrettlowe/esp8266/libraries//Adafruit-BMP085-Library/
/Users/barrettlowe/esp8266/libraries//Adafruit_INA219/
/Users/barrettlowe/esp8266/libraries//Adafruit_MLX90614_Library/
/Users/barrettlowe/esp8266/libraries//Adafruit_Unified_Sensor/
/Users/barrettlowe/esp8266/libraries//ArduinoOTA/
/Users/barrettlowe/esp8266/libraries//AsyncDelay/src/
/Users/barrettlowe/esp8266/libraries//BMP180_Breakout_Arduino_Library/src/
/Users/barrettlowe/esp8266/libraries//DHT_sensor_library/
/Users/barrettlowe/esp8266/libraries//DNSServer/src/
/Users/barrettlowe/esp8266/libraries//EDB/
/Users/barrettlowe/esp8266/libraries//EEPROM/
/Users/barrettlowe/esp8266/libraries//ESP8266AVRISP/src/
/Users/barrettlowe/esp8266/libraries//ESP8266HTTPClient/src/
/Users/barrettlowe/esp8266/libraries//ESP8266HTTPUpdateServer/src/
/Users/barrettlowe/esp8266/libraries//ESP8266LLMNR/
/Users/barrettlowe/esp8266/libraries//ESP8266NetBIOS/
/Users/barrettlowe/esp8266/libraries//ESP8266SSDP/
/Users/barrettlowe/esp8266/libraries//ESP8266WebServer/src/
/Users/barrettlowe/esp8266/libraries//ESP8266WebServer/src/detail/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/include/
/Users/barrettlowe/esp8266/libraries//ESP8266WiFiMesh/src/
/Users/barrettlowe/esp8266/libraries//ESP8266httpUpdate/src/
/Users/barrettlowe/esp8266/libraries//ESP8266mDNS/
/Users/barrettlowe/esp8266/libraries//Ethernet/examples/udpntpclient_pedantic/
/Users/barrettlowe/esp8266/libraries//Ethernet/src/
/Users/barrettlowe/esp8266/libraries//Ethernet/src/utility/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/internal/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/xtensa/
/Users/barrettlowe/esp8266/libraries//GDBStub/src/xtensa/config/
/Users/barrettlowe/esp8266/libraries//Hash/src/
/Users/barrettlowe/esp8266/libraries//Hash/src/sha1/
/Users/barrettlowe/esp8266/libraries//MemoryFree/
/Users/barrettlowe/esp8266/libraries//SD/src/
/Users/barrettlowe/esp8266/libraries//SD/src/utility/
/Users/barrettlowe/esp8266/libraries//SPI/
/Users/barrettlowe/esp8266/libraries//SPISlave/src/
/Users/barrettlowe/esp8266/libraries//Servo/src/
/Users/barrettlowe/esp8266/libraries//Servo/src/esp8266/
/Users/barrettlowe/esp8266/libraries//SoftWire/src/
/Users/barrettlowe/esp8266/libraries//TFT_Touch_Shield_V2/
/Users/barrettlowe/esp8266/libraries//Ticker/
/Users/barrettlowe/esp8266/libraries//Wire/
/Users/barrettlowe/esp8266/libraries//esp8266/src/
/Users/barrettlowe/esp8266/libraries/BMP180_Breakout_Arduino_Library//src/
/Users/barrettlowe/esp8266/libraries/ESP8266HTTPUpdateServer//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WebServer//src/detail/
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi//src/
/Users/barrettlowe/esp8266/libraries/ESP8266WiFi//src/include/
/Users/barrettlowe/esp8266/libraries/SoftWire//src/
* Library source files:
LoweWeather.cpp
.//ESP_httplib.cpp
.//LoweWeather.cpp
.//weatherCalcs.cpp
/Users/barrettlowe/esp8266/libraries/Adafruit_MLX90614_Library/Adafruit_MLX90614.cpp
/Users/barrettlowe/esp8266/libraries/BMP180_Breakout_Arduino_Library/src/SFE_BMP180.cpp
/Users/barrettlowe/esp8266/libraries/DHT_sensor_library/DHT.cpp
/Users/barrettlowe/esp8266/libraries/DHT_sensor_library/DHT_U.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer.cpp
/Users/barrettlowe/esp8266/libraries/ESP8266mDNS/ESP8266mDNS.cpp
[... 141 lines cut ...]
"/Users/barrettlowe/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-I/Users/barrettlowe/esp8266/tools/sdk/include" "-I/Users/barrettlowe/esp8266/tools/sdk/lwip/include" "-I/Users/barrettlowe/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/mkESP/LoweWeather_nodemcuv2/core"  -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10605 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_esp8266 -DARDUINO_BOARD=\"ESP8266_NODEMCU\"  -DESP8266  -I/Users/barrettlowe/esp8266/cores/esp8266 -I/Users/barrettlowe/esp8266/variants/nodemcu -I/tmp/mkESP/LoweWeather_nodemcuv2 -I.// -I/Users/barrettlowe/esp8266/libraries//Adafruit-BMP085-Library/ -I/Users/barrettlowe/esp8266/libraries//Adafruit_INA219/ -I/Users/barrettlowe/esp8266/libraries//Adafruit_MLX90614_Library/ -I/Users/barrettlowe/esp8266/libraries//Adafruit_Unified_Sensor/ -I/Users/barrettlowe/esp8266/libraries//ArduinoOTA/ -I/Users/barrettlowe/esp8266/libraries//AsyncDelay/src/ -I/Users/barrettlowe/esp8266/libraries//BMP180_Breakout_Arduino_Library/src/ -I/Users/barrettlowe/esp8266/libraries//DHT_sensor_library/ -I/Users/barrettlowe/esp8266/libraries//DNSServer/src/ -I/Users/barrettlowe/esp8266/libraries//EDB/ -I/Users/barrettlowe/esp8266/libraries//EEPROM/ -I/Users/barrettlowe/esp8266/libraries//ESP8266AVRISP/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266HTTPClient/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266HTTPUpdateServer/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266LLMNR/ -I/Users/barrettlowe/esp8266/libraries//ESP8266NetBIOS/  [... cut ...] -I/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266WiFi/src/include/ -I/Users/barrettlowe/esp8266/libraries//ESP8266WiFiMesh/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266httpUpdate/src/ -I/Users/barrettlowe/esp8266/libraries//ESP8266mDNS/ -I/Users/barrettlowe/esp8266/libraries//Ethernet/examples/udpntpclient_pedantic/ -I/Users/barrettlowe/esp8266/libraries//Ethernet/src/ -I/Users/barrettlowe/esp8266/libraries//Ethernet/src/utility/ -I/U

plerup, 2017-07-23:
cygwin isn't really my turf, but please make sure that the character case is correct for the full path of the reported missing file

BarrettLowe, 2017-08-04:
I ignored this for a while but now it's really bugging me. I'm not using cygwin. When I run make list_libs, all necessary libraries are there. Spelled correctly and correct case. I've got to be missing something else or there is a bug somewhere...

BarrettLowe, 2017-08-26:
For the record, that is not really the right fix - am I really the only one with this problem? I've tried loading short example sketches that include a library .h file and even those fail. Hopefully someone has an idea of what I've got wrong in my setup. Thanks

plerup, 2017-08-27:
Which OS are you on?

Please provide the complete short example that you now have failing

BarrettLowe, 2017-08-29:
Take the following file named TempSketch.ino
```
#include <Adafruit_INA219.h>
void setup(){}
void loop(){}
```

My $ESP_ROOT is set to /Users/barrettlowe/esp8266.
The Adafruit_INA219.h file is located within /Users/barrettlowe/esp8266/libraries.
The command I run is
`make -f /Users/barrettlowe/Downloads/makeEspArduino/makeEspArduino.mk SKETCH=TempSketch.ino BOARD=nodemcuv2
`

This will fail with the message the the target cannot be found for Adafruit_INA219.cpp needed by Adafruit_INA219.o.

In prepping for this post, I gained more information. In the same directory as TempSketch.ino, there are 4 other files - 2 cpp files and their corresponding headers. REMOVING those files, allows this to build successfully - WHY? 

Example tree (error)
-/User/barrettlowe/Documents
|--TempSketch
|----TempSketch.ino
|----AClass.h
|----AClass.cpp
|----AnotherClass.h
|----AnotherClass.cpp

Example tree (successful)
-/User/barrettlowe/Documents
|--TempSketch
|----TempSketch.ino

Is this behavior expected? If I need to use AClass and AnotherClass where should those files be stored?

plerup, 2017-08-29:
Ok, this may be a Mac thing. At least the initial constant "Build state has changed, doing a full rebuild" and bash /proc warning is.
I am now testing on a Mac but haven't been able to recreate your error. Could you just create a zip or similar with your small tree?

everslick, 2017-11-12:
The 'bash /proc' warning should be fixed by now (see #56). users on MacOS gave me trouble due to its case insensitive filesystem (fill in your favorite swear word here).

closed 2017-11-20 by plerup

dinhduongha, 2018-10-18:
@BarrettLowe @plerup 
I also have this problem within MacOS.
I debug and see: whenever use any libraries ( except $(ESP_ROOT)/cores/esp32), $(USER_SRC) and $(USER_OBJ) are correct, but compiler can't find libraries's source files to compile to .obj files.
For ex: If you use EEPROM library, compiler can't find $(ESP_ROOT)/libraries/EEPROM/EEPROM.cpp
I fixed (dirty) at my repos: [ https://github.com/dinhduongha/makeEspArduino ](https://github.com/dinhduongha/makeEspArduino).