plerup/makeEspArduino #43: finding libraries for project that compiles on the IDE

foice, 2017-07-08:
Hi, I am able to build and flash simple sketches (such as the "Blink") using a `espmake`. However I have trouble with projects of my own that use more libraries and that I can successfully build and flash with the IDE (I am on Raspbian if that matters).

Now I am stuck on some missing libraries (see below). I have this library in 
`/home/pi/bin/./arduino-1.8.3/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.h`
Any idea how to fix my makefile? 

More in general would be nice to read ideas on how to derive a working makefile from the IDE compilation.

     `In file included from /home/pi/Arduino/libraries/Adafruit_MQTT_Library/examples/mqtt_fona/fonahelper.cpp:1:0:`
   `/home/pi/Arduino/libraries/Adafruit_SleepyDog_Library/Adafruit_SleepyDog.h:23:4: error: #error         Unsupported platform for the Adafruit Watchdog library!`
      `     #error Unsupported platform for the Adafruit Watchdog library!
      ^`
     `/home/pi/Arduino/libraries/Adafruit_MQTT_Library/examples/mqtt_fona/fonahelper.cpp:2:28: fatal   error: SoftwareSerial.h: No such file or directory`
    ` #include <SoftwareSerial.h>`
                            
    `compilation terminated.`
    `/home/pi/makeEspArduino/makeEspArduino.mk:194: recipe for target`
    `'/tmp/mkESP/dht_temp_generic/fonahelper.cpp.o' failed`
    ` make: *** [/tmp/mkESP/dht_temp_generic/fonahelper.cpp.o] Error 1`  
`make: *** Waiting for unfinished jobs....`

plerup, 2017-07-11:
You can not use the SoftwareSerial for AVR when building for esp8266

Also if you want to use libraries outside the standard directories you have to specify them via LIBS variable

foice, 2017-07-11:
Hi, 
  at the moment my project includes
```
#include <ESP8266WiFi.h>
#include <Adafruit_MQTT.h>
#include <Adafruit_MQTT_Client.h>
#include <DHT.h>
#include <DHT_U.h>
```

are you saying that some of these libraries cannot be used with makeEspArduino? do you see any workaround? (for instance substitute libraries)

plerup, 2017-07-12:
Well, there is a SoftwareSerial for esp8266, written by yours truly: https://github.com/plerup/espsoftwareserial

foice, 2017-07-13:
So it sounds like you are suggesting to clone this and it to the LIBS path
of my current makefile?

foice, 2017-07-13:
Ok, I did clone it (in `~/Arduino/libraries`so espmake did find it without any change to the makefile) and I think we made one line of progress, because now the compilation stops one line below where it stopped before in `fonahelper.cpp`

```
In file included from /home/pi/Arduino/libraries/Adafruit_MQTT_Library/examples/mqtt_fona/fonahelper.cpp:1:0:
/home/pi/Arduino/libraries/Adafruit_SleepyDog_Library/Adafruit_SleepyDog.h:23:4: error: #error Unsupported platform for the Adafruit Watchdog library!
   #error Unsupported platform for the Adafruit Watchdog library!
    ^
/home/pi/Arduino/libraries/Adafruit_MQTT_Library/examples/mqtt_fona/fonahelper.cpp:3:27: fatal error: Adafruit_FONA.h: No such file or directory
 #include "Adafruit_FONA.h"
                           ^
compilation terminated.
/home/pi/makeEspArduino/makeEspArduino.mk:194: recipe for target '/tmp/mkESP/dht_temp_generic/fonahelper.cpp.o' failed
```
I am also worried about the `#error Unsupported platform for the Adafruit Watchdog library!`, is that as bad as it sounds? 

Anyhow I wanted to understand what is the fundamental issue here, seems to be that stuff I can compile with the IDE does not necessarily compile with espmake, because some libraries cannot be used (or do not exist) for espmake. Am I getting it right?

In that case do you have any suggestion to replace these libraries with ESPmake-able alternatives?

Thanks for advising

plerup, 2017-07-20:
Well, don't know the details about these libraries but the AdaFruit Sleepy Dog library doesn't list esp8266 among the supported boards

closed 2017-11-20 by plerup

JarrettR, 2018-09-06:
Just to solve this for future searches:

You're getting this because Make is trying to recursively compile all the examples when you've added the whole repo in your library search path. Just delete the examples.

`rm -rf Adafruit_MQTT_Library/examples/`