plerup/makeEspArduino #44: Linker error with a multi-file project

mat-cab, 2017-07-16:
Hi,
I have a project which I try to build using makeEspArduino Makefile, and I seem to run into some issues when using multiple files in my project (to create several libraries for the project for example).
More particularly, I have build an independent "Wifi_Lib" library, in separate files (Wifi_Lib.h & Wifi_Lib.cpp). in Wifi_Lib.cpp, the appropriate include is executed:

```#include "Arduino.h"
#include "ESP8266WiFi.h"
WiFiClient client;
...
```

This library is included in the main ino file of the project.
When linking I have the following issue:

```Linking /ESP8266-Daikin-Remote.bin
  Versions: 815e3a0-dirty, 2.3.0
/Wifi_Lib.cpp.o:(.text._Z14initializeWifiv+0x0): undefined reference to `WiFi'
/Wifi_Lib.cpp.o:(.text._Z14initializeWifiv+0x4): undefined reference to `ESP8266WiFiGenericClass::mode(WiFiMode)'
/Wifi_Lib.cpp.o: In function `initializeWifi()':
/Wifi_Lib.cpp:214: undefined reference to `ESP8266WiFiGenericClass::mode(WiFiMode)'
```

An include of ``ESP8266WiFi.h`` in the main ino file solves the issue. However, this include is not mandatory per se in the ino file., only in the library.
I think the issue is related to the automatic search performed for LIBS (which is only executed for the sketch file, not for the dependent files).

Is there any possible fix for this ? Or are there some guiding rules for a multifile project with makeEspArduino ?

Thank you,

plerup, 2017-07-16:
In this case you have to specify the LIBS variable yourself, either on the command line or in a config.mk
The automatic search is not recursive, for now anyway

mat-cab, 2018-02-27:
Closed after update to 4.10.0 (see #45)

closed 2018-02-27 by mat-cab