plerup/makeEspArduino #47: Question: How to set CPU frequency

aj-ptw, 2017-07-26:
Hi I am trying to set the cpu frequency to 160MHz instead of 80MHz. 

Will setting `F_CPU` to `160000000l` set the speed? The program uploads find but i'm not seeing anyway to verify the speed was actually set.

`huzzah` for `BOARD` defaults to 80MHz

```mk
# My makefile
LIBARAIES_DIR = $(HOME)/Documents/Arduino/libraries
SKETCH = $(LIBARAIES_DIR)/OpenBCI_Wifi/examples/DefaultBoard/DefaultBoard.ino
OPENBCI_WIFI_DIR = $(LIBARAIES_DIR)/OpenBCI_Wifi

UPLOAD_PORT = /dev/cu.usbserial-A104JV88
ESP_ROOT = $(HOME)/esp8266
ESP_LIBS = $(ESP_ROOT)/libraries

BOARD = huzzah

FLASH_DEF = 4M1M
F_CPU = 160000000l

LIBS = $(ESP_LIBS)/Wire $(ESP_LIBS)/ESP8266WiFi  $(ESP_LIBS)/DNSServer $(ESP_LIBS)/ESP8266SSDP $(ESP_LIBS)/ESP8266WebServer $(LIBARAIES_DIR)/WiFiManager $(ESP_LIBS)/SPISlave $(LIBARAIES_DIR)/ArduinoJson $(ESP_LIBS)/ESP8266mDNS $(ESP_LIBS)/ESP8266HTTPUpdateServer $(ESP_LIBS)/ArduinoOTA $(LIBARAIES_DIR)/PubSubClient $(LIBARAIES_DIR)/WebSockets $(ESP_LIBS)/Hash

EXCLUDE_DIRS = $(LIBARAIES_DIR)/ArduinoJson/test $(LIBARAIES_DIR)/ArduinoJson/third-party $(LIBARAIES_DIR)/ArduinoJson/fuzzing $(LIBARAIES_DIR)/WebSockets/examples $(LIBARAIES_DIR)/PubSubClient/tests

include $(HOME)/makeEspArduino/makeEspArduino.mk
```

plerup, 2017-07-26:
Yes, this variable is used as a define in all compile operations. You can use the option VERBOSE=1 to verify this

aj-ptw, 2017-07-26:
So COOL! Thanks! I verified with VERBOSE=1 

You are awesome!

closed 2017-07-26 by aj-ptw