plerup/makeEspArduino #48: Handling Multiple .ino files

eroom1966, 2017-07-30:
Firstly, can I say great work on building this infrastructure :-)

In my usage I have multiple .ino files, and .h files, I am not sure how I can handle this, it seems you are expecting a single .ino file.
Is there a way to handle what I need ?

Kind Regards

plerup, 2017-08-01:
Good point, ino is currently not supported as libs but can easily be added. Will happen in the next commit

everslick, 2017-11-12:
for now simply renaming your *.ino files to *.cpp should do the trick. I only leave one .ino file in my source directory i.e. $(PROJECT).ino

plerup, 2018-04-08:
The problem with including all .ino files would be that any examples in the libraries will be added as well in the wildcard searches

closed 2018-04-08 by plerup

mgouget, 2018-08-19:
Hello plerup, and thanks for what looks like the most advanced system for using seriously the arduino environment!

My app has a main .ino file (walev.ino), and a second .ino with all the support functions (esp8266plus.ino). 
The second module starts with:
`#include` "ESP8266WiFi.h"
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <ESP8266HTTPUpdateServer.h>
#include <EEPROM.h>`

If I rename it to esp8266.cpp, I get the following:

`~/esp8266/walev$ make ota
walev.ino
esp8266plus.cpp
In file included from esp8266plus.cpp:3:0:
esp8266plus.h:6:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include "ESP8266WiFi.h"
                         ^
compilation terminated.
/home/mgouget/makeEspArduino/makeEspArduino.mk:227: recipe for target '/home/mgouget/Arduino/walev/.build/walev_generic/esp8266plus.cpp.o' failed
make: *** [/home/mgouget/Arduino/walev/.build/walev_generic/esp8266plus.cpp.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from walev.ino:4:0:
esp8266plus.h:6:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include "ESP8266WiFi.h"
                         ^
compilation terminated.
/home/mgouget/makeEspArduino/makeEspArduino.mk:231: recipe for target '/home/mgouget/Arduino/walev/.build/walev_generic/walev_.cpp.o' failed

I tried to symlink all the library includes in a directory that I add to 'BUILD_EXTRA_FLAGS= -I$(SKETCHDIR)/includes_8266' but it does not work:
make ota
walev.ino
esp8266plus.cpp
In file included from esp8266plus.h:6:0,
                 from esp8266plus.cpp:3:
./includes_esp8266/ESP8266WiFi.h:28:36: fatal error: include/wl_definitions.h: No such file or directory
 #include "include/wl_definitions.h"
                                    ^
compilation terminated.
/home/mgouget/makeEspArduino/makeEspArduino.mk:227: recipe for target '/home/mgouget/Arduino/walev/.build/walev_generic/esp8266plus.cpp.o' failed
make: *** [/home/mgouget/Arduino/walev/.build/walev_generic/esp8266plus.cpp.o] Error 1
make: *** Waiting for unfinished jobs....
In file included from esp8266plus.h:6:0,
                 from walev.ino:4:
./includes_esp8266/ESP8266WiFi.h:28:36: fatal error: include/wl_definitions.h: No such file or directory
 #include "include/wl_definitions.h"
                                    ^
compilation terminated.
'
Any suggestions?

Michel