plerup/makeEspArduino #52: setting ARDUINO_ARCH_

vitoller, 2017-10-08:
I am using the make file under linux (ubuntu), from a project originally set up with platformio (atom).

One of the projects uses the RingBuffer library.
    https://github.com/wizard97/ArduinoRingBuffer

The library checks the architecture:
    #elif defined(ARDUINO_ARCH_ESP8266)

with the makefile the architecture is set to:
ARDUINO_ARCH_esp8266

as a workaround I added:
BUILD_EXTRA_FLAGS = -DARDUINO_ARCH_ESP8266

Is there a better way of doing this?

plerup, 2017-10-08:
This is a bug. Will be fixed in next commit

plerup, 2017-10-10:
Done

closed 2017-10-10 by plerup

vitoller, 2017-10-12:
great! thanks for the quick fix

dopefishh, 2020-11-10:
This bug resurfaced at some point because with the latest greatest tip it is again ARDUINO_ARCH_esp8266.