plerup/makeEspArduino #56: Does not work on OSX

jackjansen, 2017-11-01:
The project doesn't work on OSX: /proc isn't available there.

I would supply a patch but I don't fully understand how the section around full rebuild works, and what the STATE_INF initialisation tries to do. For the time being in my private copy I've disabled the whole bit with ifneq ($OS, Darwin), but that doesn't seem like a real solution.

plerup, 2017-11-01:
I searched earlier for a way to get the process command line on OSX but to no avail.
You can get around this now by adding IGNORE_STATE=1 to the make command.

Unless someone else finds a solution I will change this to be the default on OSX

jackjansen, 2017-11-01:
Maybe MAKEFLAGS has enough of the information you need? See for example https://www.gnu.org/software/make/manual/html_node/Options_002fRecursion.html

plerup, 2017-11-02:
What needs to be checked are possible variable definitions on the command line and these are not passed via MAKEFLAGS

plerup, 2017-11-02:
Try the latest commit, should be fixed now

closed 2017-11-12 by plerup