plerup/makeEspArduino #57: add a $(CUSTOM_DEPENDS) for user dependecies

nemo9955, 2017-11-06:
Add an empty variable like `"$(CUSTOM_DEPENDS) $(ESP_LIBS) $(ARDUINO_LIBS)" $(SKETCH))` at the end of the `LIBS :=` initialization so a user can include his custom libraries alongside the Arduino and ESP ones.

Personally, I thought that is what LIBS was for (specifying some general libs that may be a dependency ) , but looking at the source, LIBS is for the EXACT libs to be compiled.

everslick, 2017-11-12:
I do it by providing the path, like this:
```
LIBS       = ../libs/WebServer             \
             $(ESP_LIBS)/ESPmDNS           \
             $(ESP_LIBS)/HTTPClient
 ```
Unless i misunderstand, what your goal is.

nemo9955, 2017-11-13:
I have a few sketches with different imports; I don't want to manually give it the path from every library for every sketch, since this makefile automatically searches the needed libs from the `import<*>` statement from a "pool" of libraries: ESP_LIBS and ARDUINO_LIBS.

plerup, 2017-11-13:
New variable CUSTOM_LIBS added in latest commit. Enables automatic search for library dependencies in additional directories.

closed 2017-11-20 by plerup