plerup/makeEspArduino #58: esp8266/Arduino latest master breaks converting to binary

everslick, 2017-11-12:
```
Creating core archive
Linking ../build/ESP8266/3.0.0/alpha/ger/emonio.bin
  Versions: 2.4.0-258-g9782fe8-dirty, 2.4.0-rc2-18-g303a71d
error: invalid flash mode value: -bf
../../makeEspArduino/makeEspArduino.mk:235: recipe for target '../build/ESP8266/3.0.0/alpha/ger/emonio.bin' failed
make: *** [../build/ESP8266/3.0.0/alpha/ger/emonio.bin] Error 2
```

I think it has something to do with parsing platform.txt, but i was unable to find out what really happens there.

plerup, 2017-11-12:
Hrm, the "build.f_cpu" property seems to be gone for some reason in boards.txt :unamused:

plerup, 2017-11-12:
Says in latest commit that boards.txt is auto generated by "tools/boards.txt.py". Can't find that file though

everslick, 2017-11-13:
I dug around to find some info on what is going on and found this:
https://github.com/esp8266/Arduino/issues/3582
https://github.com/esp8266/Arduino/pull/3722/commits/e07101a5df0f397aa79971a701494d3333d7e7f3

plerup, 2017-11-13:
Fixed in latest commit. Thanks for noticing

everslick, 2017-11-13:
Great! Works!!! Awesome work!

closed 2017-11-13 by everslick