plerup/makeEspArduino #59: LWIP2 support on ESP8266

everslick, 2017-11-15:
since a few days the Arduino core for ESP8266 supports the new LWIP2 tcp/ip stack. If I understand correctly this new implementation can be switched to in the Arduino IDE menue.

https://github.com/esp8266/Arduino/pull/3783/commits/4f93f7f74f2f020f6260e7e7155f27785ea15b8c

AFAICS there are 3 things necessary to link against LWIP2:

Trigger the build of the library
Add the correct include path
Add the archive to the static libs

LWIP2 should improve memory consumption and excution speed, so it could be worthwhile to add support to makeEspArduino.

plerup, 2017-11-15:
Well doesn't this work automatically? The library liblwip2.a is prebuilt and everything controlled via platform.txt is included.

Or do you mean that we should include rules for rebuilding the lwip2 library from the git submodule?

BTW, looks like it was the changes in this commit that broke the previous version of makeEspArduino :smirk:

everslick, 2017-11-15:
when compiling with VERBOSE=1 and looking at the linking stage i see:

-lhal -lphy -lpp -lnet80211 *-llwip_gcc* -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc

but the lwip2 lib is in the archive *liblwip2.a* so the I _think_ LWIP2 support is opt-in at least for now. no?

everslick, 2017-11-15:
Also the emitted include path for the headers is in my case *-I../../Arduino-ESP8266/tools/sdk/lwip/include*, but it would have to be *-I../../Arduino-ESP8266/tools/sdk/lwip2/include*.

plerup, 2017-11-15:
Ok, I guess this is intended to be controlled via the property _build.lwip_include_ for compilations but I can't find anything references in the link command.

Nevertheless once I have figured this out there will be an optional LWIP_VERSION or alike which can be used to control this in makeEspArduino

everslick, 2017-11-15:
I have seen *build.lwip_lib=-llwip_gcc*

do you mean that?

plerup, 2017-11-16:
I guess so, I have to test in Arduino IDE to check

plerup, 2017-11-19:
Fixed in latest commit. Use the variable LWIP_VARIANT, more info in the commit message
Thanks for keeping an eye on what happens in esp8266/Arduino :smiley:

everslick, 2017-11-19:
Thanks, now I can switch between different LWIP VERSION 1 variants (I tested all 4 of them), but still non of those gave me the LWIP VERSION 2.

For LWIP 2 we would need to change the LWIP include path to [..]/sdk/lwip2/include and link against -llwip2.a

Maybe we need an additional control varaible like the LWIP_VERSION=2 you mentioned above?

Thanks for your great work! Maintaining this is a never ending game of whac-a-mole! :-D

plerup, 2017-11-19:
Strange, the settings for "open" are 

generic.menu.LwIPVariant.open=v2
generic.menu.LwIPVariant.open.build.lwip_include=lwip2/include
generic.menu.LwIPVariant.open.build.lwip_lib=-llwip2

everslick, 2017-11-19:
OK, will double check. hang on...

everslick, 2017-11-19:
OH, LOL, c'mon.. seriously?

I just did not pull. sorry for the confusion! all is well! (as usual) :+1: 

thanks! closing!

closed 2017-11-19 by everslick