plerup/makeEspArduino #60: cannot make any program work on ESP32

elpaso, 2017-11-26:
Sorry for the very generic title, I was using your makefile on ESP8266 without any issue and I'm now trying it with ESP32.
I downloaded the toolchain from https://github.com/espressif/arduino-esp32.git and followed instructions for Linux (I'm xenial 64bits) and I can compile all sketches just fine. 
Flashing also shows no errors.
But even a simple blink script does not work: I don't have any output in the serial monitor (cutecom) too.
The chip seems to be ok because I've successfully flashed the nodemcu firmware and tested some simple scripts.

Is there anything I could do to diagnose the issue?

plerup, 2017-11-28:
Please try the latest commit and make sure that you use the correct BOARD= definition

elpaso, 2017-11-28:
Hi, jI've ust tried with no apparent change, btw my board is the WROOM 32 (it should be the default one).

The sketch was just a blink on pin 2 and hello world on serial port.

```
Memory usage
  Ram:    11016 bytes
  Flash: 142165 bytes

perl -e 'print "Build complete. Elapsed time: ", time()-1511905232,  " seconds\n\n"'
Build complete. Elapsed time: 0 seconds

python "/home/ale/esp32/arduino-esp32/tools/esptool.py" --chip esp32 --port "/dev/ttyUSB0" --baud 921600  --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 "/home/ale/esp32/arduino-esp32/tools/partitions/boot_app0.bin" 0x1000 "/home/ale/esp32/arduino-esp32/tools/sdk/bin/bootloader.bin" 0x10000 "/tmp/mkESP/Blink_esp32/Blink.bin" 0x8000 "/tmp/mkESP/Blink_esp32/Blink.partitions.bin"
esptool.py v2.1
Connecting........_____..
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 33554.4 kbit/s)...
Hash of data verified.
Flash params set to 0x022f
Compressed 12320 bytes to 8150...
Wrote 12320 bytes (8150 compressed) at 0x00001000 in 0.1 seconds (effective 1048.5 kbit/s)...
Hash of data verified.
Compressed 143312 bytes to 80143...
Wrote 143312 bytes (80143 compressed) at 0x00010000 in 1.1 seconds (effective 1052.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 8387.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```

From the terminal:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11404
entry 0x40078aa0
```

plerup, 2017-11-29:
I don't have a WROOM32 myself but I have seen similar problems with another board depending on wrong default flash mode being specified in the Arduino settings.
Please try different modes by defining FLASH_MODE= on the command line. Possible values are dio, dout  and qio

closed 2018-04-08 by plerup