plerup/makeEspArduino #61: Missing memory configurations for existing board causes "Unknown Board" message

sluebbert, 2017-11-26:
tldr; Notify user when $(FLASH_DEF) is empty instead of just board not found.

While messing around with pull requests in esp8266/Arduino, a changeset somewhere removed the [BOARD].menu.FlashSize section for my board in the boards.txt file.
With no flash size loaded, the call to PARSE_ARDUINO is missing the $$flashSize parameter which causes the rest of the parameters to be offset incorrectly when passed in.  In my scenario, the board.txt filename parameter got left out of the mix for actually searching for boards.  This issue gave me the "Unknown board [BOARD]" message which took me down a hole of troubleshooting the missing board that certainly existed within the boards.txt.

plerup, 2017-11-27:
Yes, saw that one. Fix is in the making

plerup, 2017-11-28:
Please try the latest commit

sluebbert, 2017-11-29:
If I wipe out all of the FlashSize options for the board, no unknown board message shows, but I do get this error after linking:
`/usr/share/arduino_esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file -Wl,--gc-sections: No such file or directory
collect2: error: ld returned 1 exit status
make: *** [/usr/share/arduino_esp8266/makeEspArduino.mk:244: /tmp/mkESP/main_thing/main.bin] Error 1`

Is there a scenario where I should be able to build with no FlashSize selected?  I won't pretend to know what is needed or not from that boards file.

All in all, much better! If FLASH_DEF is required to have content for success, it may be nice to fail fast with a message requiring it.

plerup, 2017-11-29:
Good point.
FLASH_DEF controls both the memory configuration parameters and which link-file to use.
Please note that you can supply it on the command line (or in a separate config file) via FLASH_DEF=

closed 2018-04-08 by plerup