plerup/makeEspArduino #66: How do I use BUILD_EXTRA_FLAGS ?

nemo9955, 2018-01-09:
I tried multiple variants, with ```=```, with ```#```, with ```
#define STRINGIZE(x) #x
#define STRINGIZE_VALUE_OF(x) STRINGIZE(x)``` in the ```.ino``` file and it either tells me macro not found or other errors .

.ino CODE
```
#define STRINGIZE(x) #x
#define STRINGIZE_VALUE_OF(x) STRINGIZE(x)

const char* update_username =  _UPDATE_USERNAME;
const char* update_password =  _UPDATE_PASSWORD;
const char* host =             _HOST;
const char* ssid =             _SSID;
const char* password =         _PASSWORD;
// const char* update_username =   STRINGIZE_VALUE_OF(_UPDATE_USERNAME);
// const char* update_password =   STRINGIZE_VALUE_OF(_UPDATE_PASSWORD);
// const char* host =              STRINGIZE_VALUE_OF(_HOST);
// const char* ssid =              STRINGIZE_VALUE_OF(_SSID);
// const char* password =          STRINGIZE_VALUE_OF(_PASSWORD);
const char* update_path =       "/update";
// const char* update_path =       _UPDATE_PATH;
```

Example outputs : 

```
"/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-L/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/lib" "-L/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/ld" "-L/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o "/tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.elf" -Wl,--start-group "/tmp/mkESP/OTA_local_to_ESP_d1_mini/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-L/tmp/mkESP/OTA_local_to_ESP_d1_mini"
"/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/esptool/esptool" -eo "/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/bootloaders/eboot/eboot.elf" -bo "/tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.bin" -bm dio -bf 40 -bz 4M -bs .text -bp 4096 -ec -eo "/tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.elf" -bs .irom0.text -bs .text -bs .data -bs .rodata -bc -ec
"/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size" -A "/tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.elf" | perl -e "$MEM_USAGE" "^(?:\.irom0\.text|\.text|\.data|\.rodata|)\s+([0-9]+).*" "^(?:\.data|\.rodata|\.bss)\s+([0-9]+).*"

Memory usage
  Ram:    35624 bytes
  Flash: 291447 bytes

printf "LwIPVariant: v2 Prebuilt (MSS=536)\n"
LwIPVariant: v2 Prebuilt (MSS=536)
printf "Flash size: 4M (1M SPIFFS)\n\n"
Flash size: 4M (1M SPIFFS)

perl -e 'print "Build complete. Elapsed time: ", time()-1515487954,  " seconds\n\n"'
Build complete. Elapsed time: 39 seconds

/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool -a soft_reset write_flash 0x00000 /tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.bin
make: execvp: /home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool: Permission denied
/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/makeEspArduino/makeEspArduino.mk:257: recipe for target 'flash' failed
make: *** [flash] Error 127

Warning: local() encountered an error (return code 2) while executing 'make -e -f /home/me/myproj/3rd-party/nodes-myproj/deploy-libs/makeEspArduino/makeEspArduino.mk BOARD="d1_mini" FLASH_DEF="4M1M" UPLOAD_PORT="/dev/ttyUSB0" SKETCH="/home/me/myproj/nodes-myproj/sketches/OTA_local_to_ESP.ino" ESP_ROOT="/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266" ESPTOOL_PY="/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool" CUSTOM_LIBS="/home/me/myproj/3rd-party/nodes-myproj/sketch-libs /home/me/myproj/nodes-myproj/node-lib" BUILD_EXTRA_FLAGS='
 -D_UPDATE_USERNAME=XXXX 
 -D_UPDATE_PASSWORD=xxxx 
 -D_HOST=esp8266_wu 
 -D_SSID=sssssssss 
 -D_PASSWORD=pppppppppp 
 -D_UPDATE_PATH=update '  flash '

```

```
"/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++"
 -D__ets__
 -DICACHE_FLASH -U__STRICT_ANSI__ "-I/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/include" "-I/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/lwip2/include" "-I/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266/tools/sdk/libc/xtensa-lx106-elf/include" "-I/tmp/mkESP/OTA_local_to_ESP_d1_mini/core"  -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections
 -DF_CPU=80000000L
 -DLWIP_OPEN_SRC
 -DTCP_MSS=536  
 -DARDUINO=10605
 -DARDUINO_ESP8266_WEMOS_D1MINI
 -DARDUINO_ARCH_ESP8266
 -DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"  
 -DESP8266
 -D_UPDATE_USERNAME \"XXXX\" 
[... 119 lines cut ...]
  Ram:    35624 bytes
  Flash: 291447 bytes

printf "LwIPVariant: v2 Prebuilt (MSS=536)\n"
LwIPVariant: v2 Prebuilt (MSS=536)
printf "Flash size: 4M (1M SPIFFS)\n\n"
Flash size: 4M (1M SPIFFS)

perl -e 'print "Build complete. Elapsed time: ", time()-1515488975,  " seconds\n\n"'
Build complete. Elapsed time: 38 seconds

/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool -a soft_reset write_flash 0x00000 /tmp/mkESP/OTA_local_to_ESP_d1_mini/OTA_local_to_ESP.bin
make: execvp: /home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool: Permission denied
/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/makeEspArduino/makeEspArduino.mk:257: recipe for target 'flash' failed
make: *** [flash] Error 127

Warning: local() encountered an error (return code 2) while executing 'make -e -f /home/me/myproj/3rd-party/nodes-myproj/deploy-libs/makeEspArduino/makeEspArduino.mk BOARD="d1_mini" FLASH_DEF="4M1M" UPLOAD_PORT="/dev/ttyUSB0" SKETCH="/home/me/myproj/nodes-myproj/sketches/OTA_local_to_ESP.ino" ESP_ROOT="/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266" ESPTOOL_PY="/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esptool" CUSTOM_LIBS="/home/me/myproj/3rd-party/nodes-myproj/sketch-libs /home/me/myproj/nodes-myproj/node-lib" BUILD_EXTRA_FLAGS='
 -D_UPDATE_USERNAME=\"XXXX\" 
 -D_UPDATE_PASSWORD=\"xxxx\" 
 -D_HOST=\"esp8266_wu\" 
 -D_SSID=\"sssssssss\" 
 -D_PASSWORD=\"pppppppppp\" 
 -D_UPDATE_PATH=\"update\" '  flash '

```

plerup, 2018-01-09:
Well in the first and last case the error message is related to the start of esptool, "Permission denied"

nemo9955, 2018-01-09:
Yes, but without ```BUILD_EXTRA_FLAGS ``` it works perfectly.
I think it sais the error is with esptool because of a ```"``` or `\` that breaks the arguments .

plerup, 2018-01-09:
Sending parameters with space from the command line is always problematic.

Can't you use a "config.mk" and put your definition of BUILD_EXTRA_FLAGS there?

nemo9955, 2018-01-09:
This tool, dynamically flashes the attached board with a specific sketch, this approach of passing data directly thru the build command is ideal ... a "config.mk" would be the same thing as generating a "CONSTANTS.h" and importing it (what I am doing now).

It is almost there, it passes data, but at some point here  https://github.com/plerup/makeEspArduino/blob/master/makeEspArduino.mk#L465
``` $$v{$$key} =~ s/(-D\w+=)"([^"]+)"/$$1\\"$$2\\"/g; ``` it messes things up .

plerup, 2018-01-09:
Try setting BUILD_EXTRA_FLAGS as an environment variable

nemo9955, 2018-01-10:
Got it to work using 
```
make
-f
/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/makeEspArduino/makeEspArduino.mk
BOARD="d1_mini"
FLASH_DEF="4M1M"
UPLOAD_PORT="/dev/ttyUSB0"
SKETCH="/home/me/myproj/nodes-myproj/sketches/OTA_local_to_ESP.ino"
FS_DIR="/home/me/myproj/nodes-myproj/filesystems/basic"
ESP_ROOT="/home/me/myproj/3rd-party/nodes-myproj/deploy-libs/esp8266"
CUSTOM_LIBS="/home/me/myproj/3rd-party/nodes-myproj/sketch-libs
/home/me/myproj/nodes-myproj/node-lib"
BUILD_EXTRA_FLAGS='
-D_UPDATE_USERNAME=\"XXXX\"
-D_UPDATE_PASSWORD=\"xxxx\"
-D_HOST=\"esp8266_wu\"
-D_SSID=\"sssssssss\"
-D_PASSWORD=\"pppppppppp\"
-D_UPDATE_PATH=\"/update\"
'
flash
```

The problem was indeed `esptool`, I was using the one from `https://github.com/espressif/esptool`, commented it out so espmake would use the one from Arduino and it worcked.
It only failed wen using `BUILD_EXTRA_FLAGS`.

closed 2018-04-08 by plerup