plerup/makeEspArduino #68: Fail to find ESP8266WiFi.h..

eeyrw, 2018-02-13:
I do something like this:
Open my cygwin terminal ( I use win64 ) and input:
```
cd ~
git clone https://github.com/plerup/makeEspArduino.git
git clone https://github.com/eeyrw/LcdTcp # It's my arduino project.
alias espmake="make -f ~/makeEspArduino/makeEspArduino.mk"
cd LcdTcp
espmake
```
Then...
```
which: no esptool.py in (/usr/local/bin:/usr/bin:/cygdrive/f/Perl64/site/bin:/cygdrive/f/Perl64/bin:/cygdrive/f/Windows/system32:/cygdrive/f/Windows:/cygdrive/f/Windows/System32/Wbem:/cygdrive/f/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/f/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/f/Program Files (x86)/GTK2-Runtime/bin:/cygdrive/f/WINDOWS/system32:/cygdrive/f/WINDOWS:/cygdrive/f/WINDOWS/System32/Wbem:/cygdrive/f/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/e/Git/cmd:/cygdrive/f/mingw64/bin:/cygdrive/f/Program Files/TortoiseGit/bin:/cygdrive/e/GNU Tools ARM Embedded/7 2017-q4-major/bin:/cygdrive/f/Users/yuan/AppData/Local/Microsoft/WindowsApps:/cygdrive/e/SDCC/bin:/cygdrive/e/CMake/bin:/cygdrive/d/Microsoft VS Code/bin:/usr/bin:/cygdrive/d/stm8flash:/cygdrive/e/avra-1.3.0/src)
which: no esptool.py in (/usr/local/bin:/usr/bin:/cygdrive/f/Perl64/site/bin:/cygdrive/f/Perl64/bin:/cygdrive/f/Windows/system32:/cygdrive/f/Windows:/cygdrive/f/Windows/System32/Wbem:/cygdrive/f/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/f/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/f/Program Files (x86)/GTK2-Runtime/bin:/cygdrive/f/WINDOWS/system32:/cygdrive/f/WINDOWS:/cygdrive/f/WINDOWS/System32/Wbem:/cygdrive/f/WINDOWS/System32/WindowsPowerShell/v1.0:/cygdrive/e/Git/cmd:/cygdrive/f/mingw64/bin:/cygdrive/f/Program Files/TortoiseGit/bin:/cygdrive/e/GNU Tools ARM Embedded/7 2017-q4-major/bin:/cygdrive/f/Users/yuan/AppData/Local/Microsoft/WindowsApps:/cygdrive/e/SDCC/bin:/cygdrive/e/CMake/bin:/cygdrive/d/Microsoft VS Code/bin:/usr/bin:/cygdrive/d/stm8flash:/cygdrive/e/avra-1.3.0/src)
abi.cpp
base64.cpp
cbuf.cpp
cont.S
cont_util.c
core_esp8266_eboot_command.c
core_esp8266_flash_utils.c
core_esp8266_i2s.c
core_esp8266_main.cpp
core_esp8266_noniso.c
core_esp8266_phy.c
core_esp8266_postmortem.c
core_esp8266_si2c.c
core_esp8266_timer.c
core_esp8266_wiring.c
core_esp8266_wiring_analog.c
core_esp8266_wiring_digital.c
core_esp8266_wiring_pulse.c
core_esp8266_wiring_pwm.c
core_esp8266_wiring_shift.c
debug.cpp
Esp.cpp
FS.cpp
FunctionalInterrupt.cpp
HardwareSerial.cpp
heap.c
IPAddress.cpp
cdecode.c
cencode.c
libc_replacements.c
MD5Builder.cpp
pgmspace.cpp
Print.cpp
Schedule.cpp
sntp-lwip2.c
spiffs_cache.c
spiffs_check.c
spiffs_gc.c
spiffs_hydrogen.c
spiffs_nucleus.c
spiffs_api.cpp
spiffs_hal.cpp
Stream.cpp
StreamString.cpp
time.c
Tone.cpp
uart.c
umm_malloc.c
Updater.cpp
WMath.cpp
WString.cpp
Creating core archive
LcdTcp.ino
In file included from LcdTcp.ino:2:0:
LcdTcp.h:11:25: fatal error: ESP8266WiFi.h: No such file or directory
 #include <ESP8266WiFi.h>
                         ^
compilation terminated.
make: *** [/home/yuan/makeEspArduino/makeEspArduino.mk:220：F:/cygwin64/tmp/mkESP/LcdTcp_generic/LcdTcp_.cpp.o] error 1

```
It seems that library ESP8266WiFi should be included manually ?
How can I make it works ?

plerup, 2018-02-13:
In the first case the needed dependencies are automatically detected from the includes in the sketch. This does not work recursively though and your includes are in the Lcdtcp.h, hence the missing ESP8266WiFi etc

In the second case there should be no need for copying of the libraries but you have to add the WifFiManager and the ESP8266WebServer directories to LIBS as well

eeyrw, 2018-02-13:
Thank for your rapid reply :) Finally I modify my config.mk to : 
```
THIS_DIR := $(realpath $(dir $(realpath $(lastword $(MAKEFILE_LIST)))))
ROOT := $(THIS_DIR)/..
LIBS = $(ESP_LIBS)/SPI \
  $(ESP_LIBS)/Wire \
  $(ESP_LIBS)/ESP8266WiFi \
  $(ESP_LIBS)/ESP8266WebServer \
  $(ESP_LIBS)/DNSServer 
BOARD = nodemcuv2
UPLOAD_PORT = COM7
UPLOAD_SPEED = 1500000
FLASH_DEF=4M3M
VERBOSE=1
```
I notice that a solution for auto including has been put forward in pull request https://github.com/plerup/makeEspArduino/pull/63 . Mayne it helps ?

plerup, 2018-02-25:
Please try the latest commit

fabiuz7, 2018-08-06:
I have tried the last commit, however the recursive autoinclude still remains a problem. I'm trying to compile the example Advanced.ino contained in the famous WiFiManager by tzapu (just to test this makefile). However I can't compiled it if I don't manually specify the ESP8266WiFi directory... My makefile is really simple:
`SKETCH = $(HOME)/workspace_arduino/Advanced/Advanced.ino
CHIP = esp8266
BOARD = d1_mini

include $(HOME)/makeEspArduino/makeEspArduino.mk`

This gives the following:
In file included from C:/Users/fabiu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/WiFiManager-development/src/WiFiManager.cpp:13:0:
C:/Users/fabiu/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.4.1/libraries/WiFiManager-development/src/WiFiManager.h:32:29: fatal error: ESP8266WiFi.h: No such file or directory
     #include <ESP8266WiFi.h>
                             ^
compilation terminated.

closed 2020-01-02 by plerup