plerup/makeEspArduino #70: USER_SRC doesn't search for C files

mjhammel, 2018-03-04:
This line
`USER_SRC = $(SKETCH) $(shell find $(LIBS) -name "*.cpp")`
should be
`USER_SRC = $(SKETCH) $(shell find $(LIBS) -name "*.cpp" -o -name "*.c")`
This allows libraries like [aJson](https://github.com/interactive-matter/aJson) to build correctly.

plerup, 2018-03-04:
This is not the case in the latest commit

https://github.com/plerup/makeEspArduino/blob/13831e0e8a7e40bdfe7714f44f85be31c15478e3/makeEspArduino.mk#L170

mjhammel, 2018-03-04:
Verified.  Unfortunately it has been quite some time since I pulled from this project and none of the tagged releases seems to work for me now.  They all say 
```
$ make
* Unkown board generic
make: *** No rule to make target '/home/mjhammel/src/ximba/xarduino/lightsw/bld/arduino.mk', needed by 'all'.  Stop.
```
The only configurable variable I can see that's changed is "BUILD_OUT" is now "BUILD_DIR".

plerup, 2018-03-04:
How have you defined the ESP_ROOT, explicit or via the Arduino installation?

In either case please make sure that this version is updated as well

mjhammel, 2018-03-04:
Yes, I point to the ESP8266 installation tree.  Here is my Makefile.  

```
# Build for Arduino-based, PiBox controlled
# IoT Device based on ESP8266.
#
# ----------------------------------------------------------------------
# Directory and other setup
# ----------------------------------------------------------------------
# Top of ESP8266 related utilites (Arduino, makeEspArduino, etc.)
# ESPTOP should be set in your copy of docs/arduino.sh
ESP_TOP = $(ESPTOP)

# Esp8266 Arduino git location
ESP_ROOT = $(ESP_TOP)/arduino

# makeEspArduino git location
ESP_MK_TOP = $(ESP_TOP)/makeEspArduino
ESP_MK     = $(ESP_MK_TOP)/makeEspArduino.mk

# Where are the ESP libraries stored?
ESPLIBDIR = $(ESP_ROOT)/libraries

# Output directory
BUILD_ROOT = $(GM_BUILD)
BUILD_DIR  = $(BUILD_ROOT)

# Upload speed, port and verbosity
UPLOAD_SPEED = 115200
UPLOAD_PORT  = $(shell echo $(SERIALDEV))
# UPLOAD_VERB  = -v

# Flash size for ESP-01
FLASH_SIZE = 1M

# ----------------------------------------------------------------------
#  Project setup
# ----------------------------------------------------------------------
# Which libraries are we using in our project?
# These work if they have .cpp files.  See also USER_DIRS.
LIBS =$(ESPLIBDIR)/ESP8266WiFi
LIBS+=$(ESPLIBDIR)/Wire
LIBS+=$(ESPLIBDIR)/ESP8266WebServer
LIBS+=$(ESPLIBDIR)/ESP8266mDNS
LIBS+=$(ESPLIBDIR)/WiFiManager
LIBS+=$(ESPLIBDIR)/DNSServer
LIBS+=$(ESPLIBDIR)/Timer
LIBS+=$(ESPLIBDIR)/AES
LIBS+=$(ESPLIBDIR)/aJson
LIBS+=$(ESPLIBDIR)/ESP8266-base64

# Name of our "sketch" (a .ino, aka .c, file)
SKETCH = PiBox/imlightsw.ino

# ----------------------------------------------------------------------
# Now includes the makefile that makes building with GNU Make easier.
# ----------------------------------------------------------------------
include $(ESP_MK_TOP)/makeEspArduino.mk

# This is required for libraries which provide .h files only - no .cpp files.
# USER_DIRS+=$(ESPLIBDIR)/aREST

# Updated CFLAGS
ifneq ($(NOWM),)
C_DEFINES += -DNOWM
else
ifneq ($(RESET),)
C_DEFINES += -DRESET
endif
endif

# Use VERBOSE=1 to have verbose build
ifneq ($(VERBOSE),)
CPP += -v
endif

# ----------------------------------------------------------------------
# Project specific targets
# ----------------------------------------------------------------------
clobber:
	rm -rf $(BUILD_ROOT)

showconfig:
	echo "ESP_TOP     : $(ESP_TOP)"
	echo "ESP_ROOT    : $(ESP_ROOT)"
	echo "SDK_ROOT    : $(SDK_ROOT)"
	echo "BUILD_ROOT  : $(BUILD_ROOT)"
	echo "INCLUDE_DIRS: $(INCLUDE_DIRS)"
	echo "BUILD_ROOT  : $(BUILD_ROOT)"
	echo "UPLOAD_SPEED: $(UPLOAD_SPEED)"
	echo "UPLOAD_PORT : $(UPLOAD_PORT)"
	echo "FLASH_SIZE  : $(FLASH_SIZE)"
	echo "ESPLIBDIR   : $(ESPLIBDIR)"
	echo "C_INCLUDES  : $(C_INCLUDES)"
	echo "USER_DIRS   : $(USER_DIRS)"
```

I use [cdtools](https://gitlab.com/cdtools/cdtools), which is my way of bouncing around different project directories.  It sets up the basic environment for me.  ESPTOP is set in my environment as such:

```
ESPTOP: /home/mjhammel/src/ximba/arduino/esp8266
```

Under $ESP_ROOT (aka $ESPTOP/arduino):

```
ls -l /home/mjhammel/src/ximba/arduino/esp8266/arduino/
total 136
-rw-rw-r--  1 mjhammel mjhammel   886 Mar  1 10:22 ISSUE_TEMPLATE.md
-rw-rw-r--  1 mjhammel mjhammel 26529 Feb 28 22:46 LICENSE
-rw-rw-r--  1 mjhammel mjhammel  5867 Mar  1 10:35 README.md
-rw-rw-r--  1 mjhammel mjhammel 53736 Mar  1 10:35 boards.txt
drwxrwxr-x  3 mjhammel mjhammel  4096 Feb 28 22:46 bootloaders
drwxrwxr-x  3 mjhammel mjhammel  4096 Feb 28 22:46 cores
drwxrwxr-x  5 mjhammel mjhammel  4096 Mar  1 10:35 doc
drwxrwxr-x 31 mjhammel mjhammel  4096 Mar  3 22:09 libraries
drwxrwxr-x  2 mjhammel mjhammel  4096 Mar  1 10:35 package
-rw-rw-r--  1 mjhammel mjhammel  6204 Mar  1 10:35 platform.txt
-rw-rw-r--  1 mjhammel mjhammel     0 Feb 28 22:46 programmers.txt
drwxrwxr-x  5 mjhammel mjhammel  4096 Mar  1 10:35 tests
drwxrwxr-x  7 mjhammel mjhammel  4096 Mar  1 10:35 tools
drwxrwxr-x 15 mjhammel mjhammel  4096 Mar  1 10:35 variants
```

And that directory's git status:

```
$ git status 
HEAD detached at 2.1.0
Untracked files:
  (use "git add <file>..." to include in what will be committed)

	libraries/AES/
	libraries/ArduinoJson/
	libraries/ESP8266-base64/
	libraries/Timer/
	libraries/WiFiManager/
	libraries/aJson/
	libraries/aREST/
	tools/sdk/lib/liblwip_src.a
	tools/sdk/lwip/

nothing added to commit but untracked files present (use "git add" to track)
```

mjhammel, 2018-03-04:
FYI: I found the commit that was last working for me (it's very old): d13cf13602f2acc9be3baa3315911a4c037dc62d

plerup, 2018-03-05:
Well, 2.1.0 is a quite old version of esp8266-arduino. Current version is 2.4.0

However this error message occurs when the line starting with "generic.name" is missing in the file boards.txt in the esp-arduino root directory. Could you please verify that it exists?

closed 2020-01-01 by plerup