plerup/makeEspArduino #74: State change detection is too aggressive

everslick, 2018-04-14:
In my case I find makeEspArduino is rebuilding the whole tree way too often (always?), so I would like to be able to switch off the state change detection altogether. I propose to simply change line

https://github.com/plerup/makeEspArduino/blob/master/makeEspArduino.mk#L192

to `IGNORE_STATE ?= $(if $(filter $(MAKECMDGOALS), help clean dump_flash restore_flash list_boards),1,)`

`?=` instead of `:=`

plerup, 2018-04-15:
Ok, but do you have an example when it triggered and shouldn't?

everslick, 2018-04-15:
For me it triggers too often, because:

* I regularly change a Makefile in a way, so I KNOW a rebuild is not necessary.
* I change the BUILD path dependent on different features I switch on on the command line, so no need to rebuild
* I sometime build without a target and then with target flash or upload

I don't consider this feature buggy per se, but just not helpful in all cases.

plerup, 2018-04-15:
Ok, the intent here is that better with one rebuild to many than one to little :wink: 
But you are right, it should be optional, will come in next commit

everslick, 2018-04-15:
> Ok, the intent here is that better with one rebuild to many than one to little

hehe. If you get paid for waiting for the build, then this would even be a real killer feature! ;-)

> But you are right, it should be optional, will come in next commit

THX!

plerup, 2018-05-14:
Done

everslick, 2018-07-23:
great work!!!

closed 2018-07-23 by everslick