plerup/makeEspArduino #76: Should there be a way to use custom partition tables?

everslick, 2018-04-19:
On ESP32 the flash is partitioned according to a file that contains the flash layout: `tools/partitions/default.bin`. The bin file has to be generated by processing a CSV input file with `gen_esp32part.py`. It would be extremely nice, if we could have a custom partition file in the (user provided) source code, that would be processed to create the bin file and then used during upload.

Or is this somehow possible now?

plerup, 2018-04-19:
Well, this is currently taken as is from the Adrduino platform.txt file.

You are right however, it would be a good idea to enable overriding this with a variable. Will be fixed.

plerup, 2018-05-14:
Fixed in latest commit

closed 2018-07-23 by plerup