plerup/makeEspArduino #78: unable to compile on windows

sanu-krishnan, 2018-04-26:
I tried the make command on windows 
`make -f makeEspArduino.mk ESP_ROOT=esp CHIP=esp32 DEMO=1` and got following error !

```
/usr/bin/sh: perl: command not found
/usr/bin/sh: perl: command not found
/usr/bin/sh: /proc/1/cmdline: No such file or directory
FIND: Parameter format not correct
/usr/bin/sh: perl: command not found
/usr/bin/sh: perl: command not found
/usr/bin/sh: perl: command not found
make: *** No rule to make target `C:/cygwin/tmp/mkESP/WiFiScan_esp32/arduino.mk', needed by `all'.  Stop.

```

the same procedure works fine in Ubuntu, and I am able to download the code to esp32 board as well. What is the mistake I did in windows? 

[image]

plerup, 2018-04-26:
You need to install Perl in your cygwin setup

sanu-krishnan, 2018-04-26:
That works. Thank you

closed 2018-04-26 by sanu-krishnan