plerup/makeEspArduino #79: eagle.app.v6.common.ld file missing.

aogdb, 2018-05-04:
hello, 

a few days ago, the auto generated file `esp8266/tools/sdk/ld/eagle.app.v6.common.ld` has been removed from project `esp8266/Arduino` (see https://github.com/esp8266/Arduino/pull/4618).

`makeEspArduino.mk` does not seem to regenerate it since my compilation fails:

```
make -f ~/makeEspArduino/makeEspArduino.mk ESP_ROOT=~/esp8266 BOARD=d1_mini flash
Linking /tmp/mkESP/test1_d1_mini/test1.bin
  Versions: Unknown, 2.4.1-82-gbd87970
/home/toto/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file ../ld/eagle.app.v6.common.ld: No such file or directory
collect2: error: ld returned 1 exit status
/home/toto/makeEspArduino/makeEspArduino.mk:252: recipe for target '/tmp/mkESP/test1_d1_mini/test1.bin' failed
make: *** [/tmp/mkESP/test1_d1_mini/test1.bin] Error 1

```

A solution ?

Eszartek, 2018-05-06:
I just ran into this issue as well. I'm not much of a Makefile jockey, but, if you run these lines before your make command, you'll be good to go:

```
ESP_ROOT=../hardware/esp8266com/esp8266
BUILD_EXTRA_FLAGS="-DNDEBUG -DVTABLES_IN_FLASH"
$ESP_ROOT/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -CC -E -P $BUILD_EXTRA_FLAGS $ESP_ROOT/tools/sdk/ld/eagle.app.v6.common.ld.h -o $ESP_ROOT/tools/sdk/ld/eagle.app.v6.common.ld
```

adjust  ESP_ROOT according to your config.mk, and modify the BUILD_EXTRA_FLAGS to reference placing the vtables into FLASH, IRAM or DRAM depending on your preference.

aogdb, 2018-05-06:
Thank you Eszartek, your command gcc works well and avoids me to install all the arduino IDE ;-)

Have a good day !

closed 2018-05-06 by aogdb