plerup/makeEspArduino #84: remove/replace rm -rf lines

paidforby, 2018-07-22:
As documented here, https://github.com/sudomesh/disaster-radio/issues/14, makeEspArduino.mk contains a line that can error into a state that will delete all files that your user has permission to access. The offending line appears to be [200](https://github.com/plerup/makeEspArduino/blob/master/makeEspArduino.mk#L200).

It's hard for me to tell what is causing the error state to occur, though my theory is that after failing once due to not defining `BUILD_DIR` or not having proper permissions to the default directory `/tmp/mkESP`, `BUILD_DIR` is set to an empty string and it enters into the `IGNORE_STATE` condition. 

Removing line 200, results in the following output,
```
cat: /tmp/mkESP/firmware_d1_mini: Is a directory
cat: /state.txt: No such file or directory
* Build state has changed, doing a full rebuild *
/bin/sh: 1: cannot create /tmp/mkESP/firmware_d1_mini: Is a directory
make: *** /tmp/mkESP/firmware_d1_mini: Is a directory.  Stop.
```

Generally, the lines that contain `rm -rf` seem dangerous and could either contain more/better checks or be replaced with less destructive, safer alternatives. Would love to know how best to fix this bug!

plerup, 2018-07-23:
Yes, this rm statement should be guarded that's true and it will be fixed in next version.
However BUILD_DIR should always get a valid default unless it is defined.

As to your problem, what is BUILD_DIR set to?

paidforby, 2018-07-23:
BUILD_ROOT was set to the default `/tmp/mkEsp` so BUILD_DIR was `/tmp/mkEsp/firmware_d1_mini`. 
After further investigation, I think I found the issue.
If I replace line 200 with,
```
$(info * build_dir = $(BUILD_DIR) but state_log = $(STATE_LOG) *)
```
I receive the output, `* build dir = /tmp/mkESP/firmware_d1_mini  but state_log = /tmp/mkESP/firmware_d1_mini /state.txt *`

At which point, the problem became obvious, we had an extra space on the BOARD definition in our settings.mk. This means that line 200 was running as, `rm -rf /tmp/mkESP/firmware_d1_mini /*`, which, needless to say, is bad. A possible solution to solution is to just drop the `/*` so there's no possibility of making this mistake. I'll promptly update our example settings. Feel free to close this issue. Thanks for the encouragement @plerup and awesome work on this makefile!

plerup, 2018-07-23:
Safer handling added in latest commit. Thanks for reporting this.

closed 2018-10-18 by plerup