plerup/makeEspArduino #85: #include <Arduino.h> in cpp files - Problem

germacovitch, 2018-07-22:
Hi there
I have a project with one .ino file and several .cpp files. I am trying to acess de serial port from on cpp file. I have included the Arduino.h in order to to that. I got serveral errors when I build the project:

/tmp/mkESP/wifi_nodemcu/ShimClient.cpp.o: In function `millis':
/home/gilberto/ESPSoftware/libraries/pubsubclient/tests/src/lib/ShimClient.cpp:8: multiple definition of `millis'
/tmp/mkESP/wifi_nodemcu/arduino.ar(core_esp8266_wiring.c.o):/home/gilberto/esp8266/cores/esp8266/core_esp8266_wiring.c:64: first defined here
/tmp/mkESP/wifi_nodemcu/arduino.ar(libc_replacements.c.o): In function `_putc_r':
/home/gilberto/esp8266/cores/esp8266/libc_replacements.c:96: multiple definition of `_putc_r'

It seems there are multiples definitions of the same object. 

Could you please advise?

BR
Gilberto

plerup, 2018-07-23:
Well, do you have a function named millis in ShimClient.cpp?

lddubeau, 2018-10-18:
I ran into the same error as the OP. That was due to using the [`pubsubclient`](https://github.com/knolleary/pubsubclient) library. I pretty much always use Arduino libraries by cloning their github repo and never ran into an issue until this one. In the `pubsubclient` repo there's a `tests` directory that ends up being compiled along with the actual library code, and causes `millis` to be defined multiple times. If I `rm -rf tests` then the error goes away.

I suspect that when people install the library through the Arduino IDE or similar development systems (e.g. PlatformIO), the `tests` subdirectory is absent.

For those of us who clone, I've not found a solution other than just removing `tests`.

plerup, 2018-10-18:
You can use the variable EXCLUDE_DIRS to exclude directories from the wildcard search

lyuzashi, 2018-10-31:
I ran into this error too where tests were being found from `pubsubclient`.
Thanks for the suggestion @plerup. To fix this in my Makefile which includes makeEspArduino, I added this line

```
EXCLUDE_DIRS=$(wildcard $(ARDUINO_LIBS)/*/tests)
```

closed 2020-01-02 by plerup