plerup/makeEspArduino #89: Linking fails with latest ESP8266 Arduino Core.

everslick, 2018-10-17:
The linker is called with an empty script, but the -T option is present:

```
"../../Arduino-ESP8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,../build/3.0.99/ahe/eng/alpha/emonio.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u app_entry -u _printf_float -u _scanf_float -Wl,-static "-L../../Arduino-ESP8266/tools/sdk/lib" "-L../../Arduino-ESP8266/tools/sdk/ld" "-L../../Arduino-ESP8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-T" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read  -o ....
```

plerup, 2018-10-17:
Are you using the latest version of makeEspArduino (4.14.2)?

everslick, 2018-10-17:
Hmm... Maybe my FLASH_DEF is not set correctly... Hang on...

plerup, 2018-10-17:
Strange, works for me (esp8266 Arduino 2.4.2-83-g64dd492) and this was one of the things fixed in latest commit. 
What happens if you do the demo build (DEMO=1)

everslick, 2018-10-18:
OK, sorry for bothering you. It really was the FLASH_DEFS variable. It was set to 1M0 instead of 1M, because this used to be the correct linker script once. I should have checked better before opening an issue. -> closing

closed 2018-10-18 by everslick