plerup/makeEspArduino #93: Problem on macOS

themodernlife, 2018-11-10:
I'm using makeEspArduino version 4.14.2 on macOS and am running into an issue running a demo project.

It gets to here:

```
Linking /tmp/mkESP/WiFiScan_esp32/WiFiScan.bin
  Versions: 1.0.0, 1.0.0
python: can't open file '/Users/ihummel/dev/example/arduino-esp32/tools/esptool/esptool.py': [Errno 2] No such file or directory
```

On macOS, it should not be using `esptool.py` but the Perl it uses to parse platform.txt to populate `ELF2BIN_COM` does this:

```
my $$os = (shift =~ /Windows_NT/) ? "windows" : "linux";
...
$$v{$$1} = $$v{$$key} if $$key =~ /(.+)\.$$os$$/;
```

So basically it will favor any OS-specific key, but in this case it will use `linux` despite having this in `platform.txt`

```
tools.esptool.cmd=esptool
tools.esptool.cmd.linux=esptool.py
tools.esptool.cmd.windows=esptool.exe
```

plerup, 2018-12-02:
Will be fixed in next commit

plerup, 2018-12-04:
Please try latest commit

closed 2019-09-15 by plerup