plerup/makeEspArduino #100: makefile wont pick any serial port speed

aguaviva, 2018-11-29:
note the '' in the output of the help command

>   CHIP                 Set to esp8266 or esp32. Default: 'esp8266'
>   BOARD                Name of the target board. Default: 'nodemcuv2'
>                          Use 'list_boards' to get list of available ones
>   FLASH_DEF            Flash partitioning info. Default '4M'
>                          Use 'list_flash_defs' to get list of available ones
>   BUILD_DIR            Directory for intermediate build files.
>                          Default '/tmp/mkESP/HelloServer_nodemcuv2'
>   BUILD_EXTRA_FLAGS    Additional parameters for the compilation commands
>   FS_DIR               File system root directory
>   UPLOAD_PORT          Serial flashing port name. Default: '/dev/ttyUSB0'
>   UPLOAD_SPEED         Serial flashing baud rate. Default: ''

aguaviva, 2018-12-02:
Just checking, do you have the latest Arduino repo?

plerup, 2018-12-04:
Was caused by changed format in Arduino files. Fixed in latest commit

closed 2019-01-09 by plerup