plerup/makeEspArduino #99: native builds

d-a-v, 2018-11-29:
There is a prelimary support to build sketches in native environment.
It would be great if your makefile could handle that in the proper arduino way.

https://github.com/esp8266/Arduino/tree/master/tests/host/README.txt#L54

everslick, 2018-12-03:
I think the point is to build a native Linux binary that can be started like an command line program. AFAIK, there are now compatible APIs that wrap standard POSIX functions for that purpose. I guess all that is needed is call the right toolchain and set some paths. I would happily test the hell out of it. ;-)

d-a-v, 2018-12-04:
Yes.
```
tests/host>make ../../libraries/ESP8266WiFi/examples/WiFiClient/WiFiClient
[...]
tests/host$ ./bin/WiFiClient/WiFiClient 
mock: TODO: wifi_set_event_handler_cb set

Connecting to your-ssid

WiFi connected
IP address: 
xx.xx.xx.xx
connecting to djxmmx.net:17
mock: TODO ClientContext::setSync()
mock: TODO setNoDelay(0)
sending data to server
receiving from remote server
"My spelling is Wobbly.  It's good spelling but it Wobbles, and the letters
 get in the wrong places." A. A. Milne (1882-1958)
closing connection
^C
```

mDNS and tcp servers are also working.

(running currently from a slightly modified version, fix will come at least for release 2.5.0)

The included Makefile is poor and does not use any kind of arduino builder, so some examples will not compile.

closed 2020-01-02 by plerup