plerup/makeEspArduino #106: Can`t compile core .cpp files

avdemeul, 2019-01-30:
Hi Peter,

First off, thanks for filling a gap I m sure many ESP aficionados were looking for! 
I m a total newbie when it comes to Makefiles so the issue I m having might be easily solved by experienced users. 

I ve installed the ESP32 libs and they have been working correctly in the Visual Micro extension for VS. 
I ve also installed Cygwin with recommended defaults + complete Perl. 
I ve added Cygwin home folder to the Windows Environment Variables Path
make and perl are both installed correctly as I get version info when running respectively make -v and perl -v in cygwin bash

Nonetheless I can t compile any sketch, not even the demo. Every time I get the same error as shown below: 
` make -f makeEspArduino.mk SKETCH=D:/ESP32/AutoHydro_v3/AutoHydro_v3.ino
/usr/bin/sh: /proc/1/cmdline: No such file or directory
FIND: Parameter format not correct
make: *** No rule to make target `D:/ESP32/AutoHydro_v3/Build/AutoHydro_v3_lolin32/base64.cpp.o', needed by `D:/ESP32/AutoHydro_v3/Build/AutoHydro_v3_lolin32/arduino.ar'.  Stop.
`

I ve tried the above not only  in CygWin but also in the Git Bash with identical results.

I ve tried to debug myself and I thought the issue must be with the rules regarding the core .cpp files but I doubt this since others seem to be able to compile without problems.
I ve been on this for a couple of days and don t know what to try next, could you shed any clarity in this matter? Any help would be greatly appreciated!

Thanks in advance!
Kind regards,
Arne

---
For the sake of completeness here are some extra output data you might need to pinpoint the issue:

**- My config.mk for the sketch**
`SKETCH = $(HOMEFOLDER)/AutoHydro_v3/AutoHydro_v3.ino
UPLOAD_PORT = COM3
UPLOAD_SPEED = 115200
BOARD = lolin32
CHIP= esp32
FLASH_DEF = 4MB
HOMEFOLDER = D:/ESP32
BUILD_ROOT = $(HOMEFOLDER)/AutoHydro_v3/Build
EXCLUDE_DIRS = $(HOMEFOLDER)/AutoHydro_v3/__vm \ $(HOMEFOLDER)/AutoHydro_v3/Release \ 
ESP_ROOT = $(HOMEFOLDER)/arduino-esp32
ESP_LIBS = $(ESP_ROOT)/libraries
LIBS =$(ESP_LIBS)/FS \
$(ESP_LIBS)/SPIFFS \
$(ESP_LIBS)/WiFi \
$(ESP_LIBS)/Wire
`

**- The partial arduino.mk file:**
`
# Board definitions
CORE_DEBUG_LEVEL ?= 0
F_CPU ?= 240000000L
FLASH_MODE ?= dio
FLASH_SPEED ?= 80m
UPLOAD_RESET ?= 
UPLOAD_SPEED ?= 921600
COMP_WARNINGS ?= -w
INCLUDE_VARIANT = lolin32
# Commands
C_COM="$(ESP_ROOT)/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H "-I$(ESP_ROOT)/tools/sdk/include/config" "-I$(ESP_ROOT)/tools/sdk/include/app_trace" "-I$(ESP_ROOT)/tools/sdk/include/app_update" "-I$(ESP_ROOT)/tools/sdk/include/asio" "-I$(ESP_ROOT)/tools/sdk/include/bootloader_support" "-I$(ESP_ROOT)/tools/sdk/include/bt" "-I$(ESP_ROOT)/tools/sdk/include/coap" "-I$(ESP_ROOT)/tools/sdk/include/console" "-I$(ESP_ROOT)/tools/sdk/include/driver" "-I$(ESP_ROOT)/tools/sdk/include/esp-tls" "-I$(ESP_ROOT)/tools/sdk/include/esp32" "-I$(ESP_ROOT)/tools/sdk/include/esp_adc_cal" "-I$(ESP_ROOT)/tools/sdk/include/esp_event" "-I$(ESP_ROOT)/tools/sdk/include/esp_http_client" "-I$(ESP_ROOT)/tools/sdk/include/esp_http_server" "-I$(ESP_ROOT)/tools/sdk/include/esp_https_ota" "-I$(ESP_ROOT)/tools/sdk/include/esp_https_server" "-I$(ESP_ROOT)/tools/sdk/include/esp_ringbuf" "-I$(ESP_ROOT)/tools/sdk/include/ethernet" "-I$(ESP_ROOT)/tools/sdk/include/expat" "-I$(ESP_ROOT)/tools/sdk/include/fatfs" "-I$(ESP_ROOT)/tools/sdk/include/freemodbus" "-I$(ESP_ROOT)/tools/sdk/include/freertos" "-I$(ESP_ROOT)/tools/sdk/include/heap" "-I$(ESP_ROOT)/tools/sdk/include/idf_test" "-I$(ESP_ROOT)/tools/sdk/include/jsmn" "-I$(ESP_ROOT)/tools/sdk/include/json" "-I$(ESP_ROOT)/tools/sdk/include/libsodium" "-I$(ESP_ROOT)/tools/sdk/include/log" "-I$(ESP_ROOT)/tools/sdk/include/lwip" "-I$(ESP_ROOT)/tools/sdk/include/mbedtls" "-I$(ESP_ROOT)/tools/sdk/include/mdns" "-I$(ESP_ROOT)/tools/sdk/include/micro-ecc" "-I$(ESP_ROOT)/tools/sdk/include/mqtt" "-I$(ESP_ROOT)/tools/sdk/include/newlib" "-I$(ESP_ROOT)/tools/sdk/include/nghttp" "-I$(ESP_ROOT)/tools/sdk/include/nvs_flash" "-I$(ESP_ROOT)/tools/sdk/include/openssl" "-I$(ESP_ROOT)/tools/sdk/include/protobuf-c" "-I$(ESP_ROOT)/tools/sdk/include/protocomm" "-I$(ESP_ROOT)/tools/sdk/include/pthread" "-I$(ESP_ROOT)/tools/sdk/include/sdmmc" "-I$(ESP_ROOT)/tools/sdk/include/smartconfig_ack" "-I$(ESP_ROOT)/tools/sdk/include/soc" "-I$(ESP_ROOT)/tools/sdk/include/spi_flash" "-I$(ESP_ROOT)/tools/sdk/include/spiffs" "-I$(ESP_ROOT)/tools/sdk/include/tcp_transport" "-I$(ESP_ROOT)/tools/sdk/include/tcpip_adapter" "-I$(ESP_ROOT)/tools/sdk/include/ulp" "-I$(ESP_ROOT)/tools/sdk/include/unity" "-I$(ESP_ROOT)/tools/sdk/include/vfs" "-I$(ESP_ROOT)/tools/sdk/include/wear_levelling" "-I$(ESP_ROOT)/tools/sdk/include/wifi_provisioning" "-I$(ESP_ROOT)/tools/sdk/include/wpa_supplicant" "-I$(ESP_ROOT)/tools/sdk/include/xtensa-debug-module" "-I$(ESP_ROOT)/tools/sdk/include/esp32-camera" "-I$(ESP_ROOT)/tools/sdk/include/esp-face" "-I$(ESP_ROOT)/tools/sdk/include/fb_gfx" $(C_PRE_PROC_FLAGS) -std=gnu99 -Os -g3 -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wpointer-arith $(COMP_WARNINGS) -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -MMD -c -DF_CPU=$(F_CPU) -DARDUINO=10605 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD=\"LOLIN32\" -DARDUINO_VARIANT=\"lolin32\"  -DESP32 -DCORE_DEBUG_LEVEL=$(CORE_DEBUG_LEVEL)  $(BUILD_EXTRA_FLAGS) $(C_INCLUDES) 
CPP_COM="$(ESP_ROOT)/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DHAVE_CONFIG_H "-I$(ESP_ROOT)/tools/sdk/include/config" "-I$(ESP_ROOT)/tools/sdk/include/app_trace" "-I$(ESP_ROOT)/tools/sdk/include/app_update" "-I$(ESP_ROOT)/tools/sdk/include/asio" "-I$(ESP_ROOT)/tools/sdk/include/bootloader_support" "-I$(ESP_ROOT)/tools/sdk/include/bt" "-I$(ESP_ROOT)/tools/sdk/include/coap" "-I$(ESP_ROOT)/tools/sdk/include/console" "-I$(ESP_ROOT)/tools/sdk/include/driver" "-I$(ESP_ROOT)/tools/sdk/include/esp-tls" "-I$(ESP_ROOT)/tools/sdk/include/esp32" "-I$(ESP_ROOT)/tools/sdk/include/esp_adc_cal" "-I$(ESP_ROOT)/tools/sdk/include/esp_event" "-I$(ESP_ROOT)/tools/sdk/include/esp_http_client" "-I$(ESP_ROOT)/tools/sdk/include/esp_http_server" "-I$(ESP_ROOT)/tools/sdk/include/esp_https_ota" "-I$(ESP_ROOT)/tools/sdk/include/esp_https_server" "-I$(ESP_ROOT)/tools/sdk/include/esp_ringbuf" "-I$(ESP_ROOT)/tools/sdk/include/ethernet" "-I$(ESP_ROOT)/tools/sdk/include/expat" "-I$(ESP_ROOT)/tools/sdk/include/fatfs" "-I$(ESP_ROOT)/tools/sdk/include/freemodbus" "-I$(ESP_ROOT)/tools/sdk/include/freertos" "-I$(ESP_ROOT)/tools/sdk/include/heap" "-I$(ESP_ROOT)/tools/sdk/include/idf_test" "-I$(ESP_ROOT)/tools/sdk/include/jsmn" "-I$(ESP_ROOT)/tools/sdk/include/json" "-I$(ESP_ROOT)/tools/sdk/include/libsodium" "-I$(ESP_ROOT)/tools/sdk/include/log" "-I$(ESP_ROOT)/tools/sdk/include/lwip" "-I$(ESP_ROOT)/tools/sdk/include/mbedtls" "-I$(ESP_ROOT)/tools/sdk/include/mdns" "-I$(ESP_ROOT)/tools/sdk/include/micro-ecc" "-I$(ESP_ROOT)/tools/sdk/include/mqtt" "-I$(ESP_ROOT)/tools/sdk/include/newlib" "-I$(ESP_ROOT)/tools/sdk/include/nghttp" "-I$(ESP_ROOT)/tools/sdk/include/nvs_flash" "-I$(ESP_ROOT)/tools/sdk/include/openssl" "-I$(ESP_ROOT)/tools/sdk/include/protobuf-c" "-I$(ESP_ROOT)/tools/sdk/include/protocomm" "-I$(ESP_ROOT)/tools/sdk/include/pthread" "-I$(ESP_ROOT)/tools/sdk/include/sdmmc" "-I$(ESP_ROOT)/tools/sdk/include/smartconfig_ack" "-I$(ESP_ROOT)/tools/sdk/include/soc" "-I$(ESP_ROOT)/tools/sdk/include/spi_flash" "-I$(ESP_ROOT)/tools/sdk/include/spiffs" "-I$(ESP_ROOT)/tools/sdk/include/tcp_transport" "-I$(ESP_ROOT)/tools/sdk/include/tcpip_adapter" "-I$(ESP_ROOT)/tools/sdk/include/ulp" "-I$(ESP_ROOT)/tools/sdk/include/unity" "-I$(ESP_ROOT)/tools/sdk/include/vfs" "-I$(ESP_ROOT)/tools/sdk/include/wear_levelling" "-I$(ESP_ROOT)/tools/sdk/include/wifi_provisioning" "-I$(ESP_ROOT)/tools/sdk/include/wpa_supplicant" "-I$(ESP_ROOT)/tools/sdk/include/xtensa-debug-module" "-I$(ESP_ROOT)/tools/sdk/include/esp32-camera" "-I$(ESP_ROOT)/tools/sdk/include/esp-face" "-I$(ESP_ROOT)/tools/sdk/include/fb_gfx" $(C_PRE_PROC_FLAGS) -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib $(COMP_WARNINGS) -Wno-error=unused-function -Wno-error=unused-but-set-var [... cut ...]
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFi.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiAP.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiClient.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiGeneric.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiMulti.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiScan.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiServer.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiSTA.cpp
D:/ESP32/arduino-esp32/libraries/WiFi/src/WiFiUdp.cpp
D:/ESP32/arduino-esp32/libraries/Wire/src/Wire.cpp
`

plerup, 2019-01-30:
Try cd to an empty directory and then

make -f makeEspArduino.mk DEMO=1

avdemeul, 2019-01-31:
Hi Peter,

Thanks for your reply.
I tried what you suggested. 
I can only get results when I cd into the folder that contains the makeEspArduino.mk file. Also when I dont define the ESP_ROOT folder it goes to look in the Arduino15 folder of LocalData but the resulting messages are identical (except for the directory), as shown below:
Please note the files and directories do exist in the ESP_ROOT folder. 

'
Arne@DESKTOP-7CIQAKK /cygdrive/d/ESP32/libraries/makeEspArduino
$ make -f makeEspArduino.mk DEMO=1 ESP_ROOT=D:/ESP32/esp32-root
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/’: No such file or directory
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/WiFi’: No such file or directory
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/WiFi’: No such file or directory
/usr/bin/sh: /proc/1/cmdline: No such file or directory
* Build state has changed, doing a full rebuild *
FIND: Parameter format not correct
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/’: No such file or directory
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/WiFi’: No such file or directory
find: ‘D:/ESP32/libraries/makeEspArduino/D:/ESP32/esp32-root/libraries/WiFi’: No such file or directory
/usr/bin/sh: /proc/1/cmdline: No such file or directory
FIND: Parameter format not correct
make: *** No rule to make target `D:/CygWin/tmp/mkESP/WiFiScan_esp32/base64.cpp.o', needed by `D:/CygWin/tmp/mkESP/WiFiScan_esp32/arduino.ar'.  Stop.
make: *** Waiting for unfinished jobs....
'

Thanks in advance!
Kind regards,
Arne

avdemeul, 2019-02-02:
Hi again Peter,

I got rid of the issue by reinstalling make via the Cygwin setup, everything is running like a charm now! Sorry for inconveniencing you and thanks again for your contribution!

Cheerio!

closed 2019-02-02 by avdemeul