plerup/makeEspArduino #107: Resulting bin uses much more ram than IDE compiled bin

polandj, 2019-02-18:
Not sure if this is related to the latest 2.5.0 ESP update, but after that update is when I noticed it. 

When I use the Arduino IDE to compile:

> Sketch uses 555448 bytes (53%) of program storage space. Maximum is 1044464 bytes.
> Global variables use 37580 bytes (45%) of dynamic memory, leaving 44340 bytes for local variables. Maximum is 81920 bytes.

When I use makeEspArduino:

> Memory usage
>   Ram:    42080 bytes
>   Flash: 556308 bytes
> 
> LwIPVariant: v2 Lower Memory

Using BOARD=nodemcuv2 and FLASH_DEF=4M2M.  Any ideas?  I hate using the IDE, but I need that ~5K of ram...

luffykesh, 2019-04-19:
I observed the same @plerup , when built from IDE:
RAM - 36588  bytes
FLASH - 360632  bytes

When build by makeEspArduino:
Memory usage
  Ram:    40648 bytes
  Flash: 360604 bytes

plerup, 2020-01-01:
Confirmed. Sorry for the long delay in responding.
This is due to Arduino IDE putting all the object files into an archive, whereas makeEspArduino sends them directly to the linker. This seems to affect the linker's ability to remove unused variables. It will be fixed in next commit, coming shortly.

plerup, 2020-01-02:
Fixed in latest commit

closed 2020-01-02 by plerup

polandj, 2020-01-02:
Hope to try it out soon.  Thanks so much!  Happy new year!