plerup/makeEspArduino #116: recipe for target 'flash' failed : pyserial usage ?

gd-git, 2019-03-17:
Hi,

since I updated the Arduino (esp8266 / Arduino) and makeEspArduino directories the make flash command no longer works:
```
$ make flash
Error in command line, need pyserial path as 1st arg and esptool path as 2nd.
/home/toto/makeEspArduino/makeEspArduino.mk:294: recipe for target 'flash' failed
make: *** [flash] Error 1
```

My Makefile that worked before.

```
ARDUINO_ROOT=$(HOME)/arduino-1.8.8
ESP_ROOT=$(HOME)/Arduino
BOARD=nodemcuv2

include $(HOME)/makeEspArduino/makeEspArduino.mk
```
Everything works with an Arduino version (esp8266 / Arduino) of Fri May 4 17:18:15 2018 and have no other information on this regression.

With my thanks.

plerup, 2019-03-17:
Please try latest commit

everslick, 2019-03-22:
upload fails here as well with latest master of makeEspArduino and Arduino core:

```
Traceback (most recent call last):
  File "/home/clemens/Devel/ESP/emonio/src/../../Arduino-ESP8266/tools/upload.py", line 25, in <module>
    esptool.main(fakeargs)
TypeError: main() takes no arguments (1 given)
make[2]: *** [/home/clemens/Devel/ESP/emonio/src/../../makeEspArduino/makeEspArduino.mk:293: flash] Error 1
```

plerup, 2019-03-22:
Are you using the latest version of this git?

And if you are using a git clone of Arduino core, have you cleaned up the mess of esptool versions by updating all submodules (update --init --recursive) and rerun get.py

everslick, 2019-03-22:
> And if you are using a git clone of Arduino core, have you cleaned up the mess of esptool versions by updating all submodules (update --init --recursive) and rerun get.py 

No, I didn't. Thx for the pointer. I will report back!

gd-git, 2019-03-22:
Hi,

sorry for the delay, everything works perfectly in my case with the latest version. Thank you because I can not do without makeEspArduino ;-)

everslick, 2019-03-22:
Confirmed working!!! :+1:

closed 2019-09-15 by plerup