plerup/makeEspArduino #118: makeEspArduino fails when ESP_ROOT has a symlink

aboulfad, 2019-04-12:
Hello,

makeEspArduino fails when `ESP_ROOT` is set to a direrctory that is a symbolic link. I use symlinks (macos) so I can switch easily between different versions of Arduino builds to do tests.

If I set `ESP_ROOT` to a dir that does NOT contain any symlinks, it works.
Example from below:

```bash
toto@mba:~/temp/test$ make -f ~/src/arduino/makeEspArduino/makeEspArduino.mk ESP_ROOT=~/src/arduino/hardware/esp8266com/esp8266
* Unknown board generic
make: *** No rule to make target `/Users/toto/temp/mkESP/test_generic/arduino.mk', needed by `all'.  Stop.
toto@mba:~/temp/test$ espmake
cont.S
make: /Users/toto/src/arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S: Permission denied
umm_malloc.cpp
```

```bash
toto@mba:~$ tree -L 1 -d ~/src/arduino/hardware/esp8266com/esp8266
/Users/toto/src/arduino/hardware/esp8266com/esp8266
├── bootloaders
├── cores
├── doc
├── libraries
├── package
├── tests
├── tools
└── variants

8 directories

toto@mba:~$ ll src/arduino/hardware/esp8266com/esp8266
lrwxr-xr-x  1 toto  staff    31B 11 Apr 12:03 src/arduino/hardware/esp8266com/esp8266@ -> /Users/toto/src/arduino/master
```

plerup, 2019-04-13:
You can easily avoid this by enclosing the definition in realpath

    make -f ~/src/arduino/makeEspArduino/makeEspArduino.mk ESP_ROOT=`realpath ~/src/arduino/hardware/esp8266com/esp8266`

aboulfad, 2019-04-13:
I just learned my 1001 *nix command ! `realpath` is not available on Macos, but `brew install coreutils` fixes that ! Thank you.

closed 2019-04-13 by aboulfad