plerup/makeEspArduino #125: Fails to find python when creating arduino.mk

matthewg42, 2019-09-08:
When building demo, I get:
```
Linking /tmp/mkESP/HelloServer_generic/HelloServer.bin                                   
  Versions: Unknown, 2.5.2                                                               
/bin/sh: /python: No such file or directory                                              
make: *** [makeEspArduino.mk:297: /tmp/mkESP/HelloServer_generic/HelloServer.bin] Error 127
```

Looking in `/tmp/mkESP/HelloServer_generic/arduino.mk` I can see it's not populating the full path to the python interpreter properly:

```
ELF2BIN_COM="/python" "$(ESP_ROOT)/tools/elf2bin.py" ... 
```

If I change it to the proper path `/usr/bin/python`, the build succeeds. How does it find any populate the python path in `arduino.mk` for the `ELF2BIN_COM` variable?  

My system is Fedora 29.

DerEros, 2019-09-13:
As a workaround, this worked for me. Not sure if it's meant to be like this. Seems that `runtime.tools.python.path` is missing, but I do not understand the workflow well enough to know, where it should really come from.

```
diff --git a/makeEspArduino.mk b/makeEspArduino.mk
index b09091e..2f06989 100644
--- a/makeEspArduino.mk
+++ b/makeEspArduino.mk
@@ -491,6 +491,7 @@ sub def_var {
    $$v{$$name} = "\$$($$var)";
 }
 
+$$v{'runtime.tools.python.path'} = '/usr/bin';
 $$v{'runtime.platform.path'} = '$$(ESP_ROOT)';
 $$v{'includes'} = '$$(C_INCLUDES)';
 $$v{'runtime.ide.version'} = '10605';
```

plerup, 2019-09-15:
Fixed in latest commit (for latest version of esp8266/Arduino)

DerEros, 2019-09-15:
Thanks for the fix. I got mixed results:

With the last release of the ESP toolchain (2.5.2), which is the one that my Arduino IDE is still downloading, it still fails. The toolchain still refers to Python2 (see https://github.com/esp8266/Arduino/blob/2.5.2/platform.txt#L111).

With the newest version from Master (2.6.0-dev) the fix works fine, no more need for a workaround.

plerup, 2019-09-15:
Fixed now, thanks for pointing this out

DerEros, 2019-09-15:
I can confirm that it now works with both toolchains.
Tried with "2.5.2" (Python 2) and "2.5.2-145-g8dd068eb" (Python 3).

Thanks! I think, the issue can be closed now.

closed 2019-09-15 by plerup