plerup/makeEspArduino #126: Partitions CSV file missing

yannS2016, 2019-10-07:
Good day !

this is my first attempt at the makefile . just following the README here. but it does not seems to compile link well. from the begining of compilations it complains about the partitions.csv that it cant find.
system: Debian stretch 64 bits
Make version: GNU Make 4.1 Built for x86_64-pc-linux-gnu

`grep: /tmp/mkESP/Blinker_esp32/partitions.csv: No such file or directory
esp32-hal-rmt.c
WMath.cpp
wiring_pulse.c
Print.cpp
esp32-hal-psram.c
esp32-hal-misc.c
wiring_shift.c
esp32-hal-dac.c
esp32-hal-time.c
stdlib_noniso.c
IPAddress.cpp
cbuf.cpp
esp32-hal-touch.c
Stream.cpp
esp32-hal-timer.c
esp32-hal-sigmadelta.c
WString.cpp
esp32-hal-i2c.c
esp32-hal-ledc.c
esp32-hal-cpu.c
main.cpp
StreamString.cpp
esp32-hal-adc.c
IPv6Address.cpp
esp32-hal-matrix.c
FunctionalInterrupt.cpp
HardwareSerial.cpp
esp32-hal-bt.c
base64.cpp
esp32-hal-spi.c
cdecode.c
cencode.c
MD5Builder.cpp
Esp.cpp
esp32-hal-uart.c
esp32-hal-gpio.c
Creating core archive
Blinker.ino
Ticker.cpp
Linking /tmp/mkESP/Blinker_esp32/Blinker.bin
  Versions: 1.0.4-8-g9ef3e2d, 1.0.4-8-g9ef3e2d
usage: gen_esp32part.py [-h] [--flash-size [{1MB,2MB,4MB,8MB,16MB}]]
                        [--disable-md5sum] [--no-verify] [--verify] [--quiet]
                        [--offset OFFSET] [--secure]
                        input [output]
gen_esp32part.py: error: argument input: can't open '/tmp/mkESP/Blinker_esp32/partitions.csv': [Errno 2] No such file or directory: '/tmp/mkESP/Blinker_esp32/partitions.csv'
/home/iotdevs/makeEspArduino/makeEspArduino.mk:291: recipe for target '/tmp/mkESP/Blinker_esp32/Blinker.bin' failed
make: *** [/tmp/mkESP/Blinker_esp32/Blinker.bin] Error 2
`
Can you help me sort this out

fijimunkii, 2019-11-22:
Same problem here. As a temp fix you can copy a partitions file to the location in the error

https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/default.csv

plerup, 2019-12-30:
Due to change of partition handling in latest commit of ESP32 Arduino.
Fix is coming shortly but in the mean time you can avoid this error by defining the variable PART_FILE and set it to the path of the default file in tools/partitions/default.csv

yannS2016, 2020-01-02:
Hello Peter1
Thanks for your answer,

i will give it a try.

plerup, 2020-01-02:
Fixed in latest commit

closed 2020-01-02 by plerup

valkiriaaquatica, 2023-09-01:
Hi, i am having that same issue right now with esp32 dev modeule, using the latest esp32 package 2.0.11 by espressif in windows 11. I have checked the platform.txt because the error was in the prebuild and iis reporting that the partitions.csv is not found even if it is in the same route where it is looking. Anyone with this problem?

valkiriaaquatica, 2023-09-03:
Finally I just isntalled it in a vmware ubuntu machine and all was fine :)