plerup/makeEspArduino #128: python3: not found (ESP8266)

everslick, 2019-10-28:
Is it only me? make insists on finding python3 in the tools directory:
```
Linking ../build/3.0.99/borabora/eng/alpha/emonio.bin
  Versions: 3.0.25-29-g5a06ffd, 2.5.2-205-g6e51ef0c
/bin/sh: 1: ../../Arduino-ESP8266/tools/python3/python3: not found
make[1]: *** [../../makeEspArduino/makeEspArduino.mk:299: ../build/3.0.99/borabora/eng/alpha/emonio.bin] Error 127
```
Makefile is up to date:
```
clemens@secrets:~/Devel/ESP$ cd makeEspArduino/
clemens@secrets:~/Devel/ESP/makeEspArduino$ git pull origin master 
From github.com:plerup/makeEspArduino
 * branch            master     -> FETCH_HEAD
Already up to date.
```
python is in $PATH:
```
clemens@secrets:~/Devel/ESP/makeEspArduino$ which python
/usr/bin/python
clemens@secrets:~/Devel/ESP/makeEspArduino$ which python3
/usr/bin/python3
```

plerup, 2019-10-28:
This looks like the problem which was fixed in the latest commit. Are you sure you are running from the git you show above?

everslick, 2019-10-28:
100% positive. I have no other repo checked out.

plerup, 2019-10-28:
Try the DEMO=1 variant

plerup, 2019-10-28:
By the way you use the IGNORE_STATE don't you? Do a clean then and build again

everslick, 2019-10-29:
yes, `IGNORE_STATE = 1`. make clean does NOT help.

plerup, 2019-10-29:
What about DEMO=1 ?

everslick, 2019-10-29:
```
Linking ../build/3.0.99/borabora/eng/alpha/HelloServer.bin
  Versions: 2.5.2-205-g6e51ef0c, 2.5.2-205-g6e51ef0c
/bin/sh: 1: ../../Arduino-ESP8266/tools/python3/python3: not found
```

plerup, 2019-10-29:
Well, then I'm lost works flawlessly here

   Linking /tmp/mkESP/HelloServer_generic/HelloServer.bin
     Versions: 2.5.2-206-g7a43092d, 2.5.2-206-g7a43092d
   Creating BIN file "/tmp/mkESP/HelloServer_generic/HelloServer.bin" using 
   "/tmp/mkESP/HelloServer_generic/HelloServer.elf"

plerup, 2019-10-29:
"git status" perhaps to make sure

everslick, 2019-10-29:
dammit! you got me. my HEAD was detached. obviously not only my git HEAD but also me real one.

sorry for wasting your time.

everslick, 2019-10-29:
what i do not understand is, that i EXPLICITLY checked out commit 1cbe5fe. more specific I do:

cd ../makeEspArduino
git pull origin master
git checkout 1cbe5fe

It seems as this does not do what I thought it would do.

everslick, 2019-10-29:
OK, before I go nuts, can you pls check that `make flash` works?

plerup, 2019-10-29:
Works here...

everslick, 2019-10-29:
thx, which OS?

plerup, 2019-10-29:
Ubuntu 18.04

plerup, 2019-10-29:
Ok, this may be an error.
I have esptool.py in my path
Investigating

plerup, 2019-10-29:
Ok, I can see that links to python is now created by ESP8266-Arduino when running the "get.py" script in the tools directory, so you must run this whenever that git is updated :see_no_evil:

everslick, 2019-10-29:
OMG, never would have guessed that. THX!

nevertheless, the fun does not stop there. it does not honor SERIAL_PORT=/dev/ttyUSBX anymore for me...
```
clemens@secrets:~/Devel/ESP/emonio-fw$ make MODEL=borabora SERIAL_PORT=/dev/ttyUSB0 usb VERBOSE=1
make flash
make[2]: Entering directory '/home/clemens/Devel/ESP/emonio-fw/src'
"../../Arduino-ESP8266/tools/python3/python3" "../../Arduino-ESP8266/tools/upload.py" --chip esp8266 --port "/dev/ttyUSB1" --baud "115200"  version --end --chip esp8266 --port "/dev/ttyUSB1" --baud "115200"  write_flash 0x0 "../build/3.0.99/borabora/eng/alpha/emonio.bin" --end
esptool.py v2.7
2.7
esptool.py v2.7
Serial port /dev/ttyUSB1
Connecting......
```

plerup, 2019-10-29:
UPLOAD_PORT is the correct variable

everslick, 2019-10-29:
oh, indeed. THX so much!

everslick, 2019-10-29:
`make erase_flash` has a similar problem :-(

```
/bin/sh: 1: /../../Arduino-ESP8266/tools/python3/python3: not found
make[1]: *** [../../makeEspArduino/makeEspArduino.mk:365: erase_flash] Error 127
```

the path now has a leading **/**

plerup, 2019-10-29:
What value are you setting for ESP_ROOT?

everslick, 2019-10-29:
ESP_ROOT = ../../Arduino-ESP8266

```
clemens@secrets:~/Devel/ESP/emonio-fw/src$ ls ../../Arduino-ESP8266
bootloaders  doc        package  tools     boards.txt         keywords.txt  package.json  POLICY.md        README.md
cores        libraries  tests    variants  ISSUE_TEMPLATE.md  LICENSE       platform.txt  programmers.txt
```

everslick, 2019-10-29:
it is the same for all build targets.

plerup, 2019-10-29:
Ok, this seems to be an error. It's a constant struggle trying to keep up with the constant changes in the arduino platform.txt 
Fix is coming

plerup, 2019-10-29:
Workaround in the mean time is to do

pip install esptool

everslick, 2019-10-29:
thx, once more!

plerup, 2019-10-30:
Ok, the handling of esptool in esp8266/Arduino is really a mess right now. I will make installation of esptool.py a requirement for some of the flashing operations. If missing an error message will be shown explaining this.

New version available now

everslick, 2019-11-12:
Thank you! :1st_place_medal:

closed 2019-11-12 by everslick