plerup/makeEspArduino #129: SPIFFS upload not working

Terstegge, 2019-11-14:
I have problems with the flash_fs on a ESP8266 target. I use FS_DIR to set the folder with the files and FLASH_DEF to partition the flash (4M1M). 
make fs works without problems, but make flash_fs tries to download the binary (HelloServer.bin in this case) and not the FS.spiffs.
Am i doing s.th. wrong??

Here is a log file:
```
andreas@nb-fh-3 ~/git/WifiDust_neu/Workspace/WifiDust $ make VERBOSE=1 DEMO=1 FS_DIR=data FLASH_DEF=4M1M flash_fs
* Build state has changed, doing a full rebuild *
perl -e "$PARSE_ARDUINO" generic '4M1M' 'Linux' 'lm2f'  /home/andreas/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/boards.txt /home/andreas/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/keywords.txt /home/andreas/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/platform.txt /home/andreas/.arduino15/packages/esp8266/hardware/esp8266/2.6.0/programmers.txt >/tmp/mkESP/HelloServer_generic/arduino.mk
echo Generating filesystem image: /tmp/mkESP/HelloServer_generic/FS.spiffs
Generating filesystem image: /tmp/mkESP/HelloServer_generic/FS.spiffs
"/home/andreas/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-4-b40a506/mkspiffs" -b 8192 -s 0xFA000 -c data /tmp/mkESP/HelloServer_generic/FS.spiffs
/jquery-3.2.1.min.js
/favicon.ico
/index.css
/Pe-icon-7-stroke.woff
/skyblue.min.css
/index.html
/index.js
/WifiDust.cfg
"/usr/bin//python3" "/home/andreas/.arduino15/packages/esp8266/hardware/
esp8266/2.6.0/tools/upload.py" --chip esp8266 --port "/dev/ttyUSB0" --baud
"115200"  flash_id --end --chip esp8266 --port "/dev/ttyUSB0" --baud "115200"
 --before default_reset --after hard_reset  write_flash 0x0
 "/tmp/mkESP/HelloServer_generic/HelloServer.bin" --end
esptool.py v2.7
Serial port /dev/ttyUSB0
Connecting........_
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:8d:ba:52
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/tmp/mkESP/HelloServer_generic/HelloServer.bin'
Makefile:349: die Regel für Ziel „flash_fs“ scheiterte
make: *** [flash_fs] Fehler 2
andreas@nb-fh-3 ~/git/WifiDust_neu/Workspace/WifiDust
```

plerup, 2019-11-14:
Please try installing esptool.py and run again

    pip install esptool

Terstegge, 2019-11-14:
Sorry, my fault. After installing esptool everything is working now! Thanks for this very handy makefile, I really like it!!

closed 2019-11-14 by Terstegge