plerup/makeEspArduino #131: Issue with building with SPIFFS

Silipwn, 2019-12-14:
On running `make -f makeEspArduino.mk flash_fs`
I get an error in the following way.

```
Generating filesystem image: /tmp/mkESP/mode_generic/FS.spiffs
PARSE ERROR: Argument: -s (--size)
             Couldn't read argument value from string '-c'
```

plerup, 2019-12-15:
You have to specify the memory configuration via FLASH_DEF
The default is 512K and no file system

Example:
make -f makeEspArduino.mk flash_fs FLASH_DEF=4M3M

Silipwn, 2019-12-15:
I tried running the example but I guess I'd have to first build the binary and then use `flash + flash_fs` option? 
I'm currently using a command similar to this:
`make -f makeEspArduino.mk flash flash_fs FLASH_DEF=4M1M`
Compilation creates a binary and uploads, but the SPIFFS functions fail.
My program fails to read the content stored.

plerup, 2019-12-15:
Do this as two separate commands, one flash and one flash_fs

Silipwn, 2019-12-16:
Doing that as separate commands, results in the below error. The `flash` command works fine and when running the `flash_fs` command, the error is seen:
```* Build state has changed, doing a full rebuild *
Generating filesystem image: /tmp/mkESP/main_generic/FS.spiffs
/data.txt
esptool.py v2.6
2.6
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/tmp/mkESP/main_generic/main.bin'
make: *** [makeEspArduino.mk:335: flash_fs] Error 2
```
I also noticed that on running `flash_fs` the /tmp/mkESP/main_generic directory is **cleared**.

closed 2019-12-16 by Silipwn