plerup/makeEspArduino #134: not able to link when SKETCH = main.cpp

paidforby, 2020-01-23:
Hi @plerup great work on this makefile!

I've noticed this issue for a while, thought I'd finally post an issue because I didn't see one that specifically addressed it. 

If I name my main sketch `main.cpp`, running `make` produces the following linker errors,
```
Linking /tmp/mkESP/main_ttgo-lora32-v1/main.bin
  Versions: 0.1.0-16-g0a569e8-dirty, 1.0.4
/tmp/mkESP/main_ttgo-lora32-v1/main.cpp.o:(.literal._Z8loopTaskPv+0x4): undefined reference to `setup()'
/tmp/mkESP/main_ttgo-lora32-v1/main.cpp.o:(.literal._Z8loopTaskPv+0x8): undefined reference to `loop()'
/tmp/mkESP/main_ttgo-lora32-v1/main.cpp.o: In function `loopTask(void*)':
/home/grant/repos/disaster-radio-refactor/./arduino-esp32/cores/esp32/main.cpp:14: undefined reference to `setup()'
/home/grant/repos/disaster-radio-refactor/./arduino-esp32/cores/esp32/main.cpp:17: undefined reference to `loop()'
collect2: error: ld returned 1 exit status
Makefile:293: recipe for target '/tmp/mkESP/main_ttgo-lora32-v1/main.bin' failed
make: *** [/tmp/mkESP/main_ttgo-lora32-v1/main.bin] Error 1
```

I know that I can solve my problem by renaming `main.cpp` to anything else, e.g. `main.ino`, `firmware.cpp`, `main-somethingelse.cpp`. I'm really just curious if you know the cause of this limitation. It would be nice to be able to name the main sketch something sensible like `main.cpp`

Fyi, the sketch I am attempting to compile is currently found in this PR, https://github.com/sudomesh/disaster-radio/pull/44/files#diff-355efca6cad1bace812bbed5026187d0

and my config.mk looks like,
```
-include $(dir $(SKETCH))settings.mk
 
SKETCH = ./firmware/esp32_ttgo/main.cpp
ESP_ROOT = ./arduino-esp32
FS_DIR = ./web/static
LIBS = ./libs/AsyncTCP \
        ./libs/ESPAsyncWebServer \
        ./libs/arduino-LoRa \
        ./libs/LoRaLayer2 \
        ./libs/AsyncSDServer \
         ./libs/TinyGPSPlus \
         ./libs/esp8266-oled-ssd1306 \
         $(ESP_LIBS)/WiFi \
         $(ESP_LIBS)/SPI \
         $(ESP_LIBS)/SD \
         $(ESP_LIBS)/ESPmDNS \
         $(ESP_LIBS)/FS \
         $(ESP_LIBS)/SPIFFS \
         $(ESP_LIBS)/Wire
BUILD_EXTRA_FLAGS='-DTTGO_LORA_V2'
```
I am attempting this on Ubuntu 18.04.

paidforby, 2020-01-23:
Note, I have tested with the latest version of makeEspArduino (4.21.0)

plerup, 2020-01-23:
The reason for this is that your main.cpp name-clashes with the real main program for the Arduino framework located in cores/esp32/main.cpp. This file contains the calls to the setup and loop procedures in your sketch. In Arduino context your sketch is not really the main program.

paidforby, 2020-01-23:
Gotcha. That makes sense. Thank you for that clarfication. So guess it is best to not name name the main sketch `main.cpp` and stick with something like "main.ino" or "sketch.cpp". Thanks!

closed 2020-01-23 by paidforby